Nitrokey/nitrokey-3-firmware #296: secrets app hangs with firmware 1.5.0 stable

eric-belhomme, 2023-06-21:
Hello,

I juste upgraded my NK3 key to latest **1.5.0** stable firmware using ` nitropy` command, and I now experiencing a crash everytime I try to **lists** secrets stored on my NK3:

```shell
rico@ricolapscc:~$ nitropy nk3 secrets list
Command line tool to interact with Nitrokey devices 0.4.38
Please provide PIN to show PIN-protected entries (if any), or press ENTER to skip
Please touch the device if it blinks
Current PIN (8 attempts left): 
Please touch the device if it blinks
^C
Aborted!
```
The nk3 LED blinked blue as usual requesting me to tap it, then the light switch to static red and nothing append, so I have to kill the command with a CTRL-C

But of course the nk3 is still hanging with red light, and doesn't respond anymore:
```shell
rico@ricolapscc:~$ nitropy nk3 status
Command line tool to interact with Nitrokey devices 0.4.38
Critical error:
An unhandled exception occurred
	Exception encountered: CtapError('CTAP error: 0x06 - CHANNEL_BUSY')

--------------------------------------------------------------------------------
Critical error occurred, exiting now
Unexpected? Is this a bug? Would you like to get support/help?
- You can report issues at: https://support.nitrokey.com/
- Writing an e-mail to support@nitrokey.com is also possible
- Please attach the log: '/tmp/nitropy.log.erzaw7pc' with any support/help request!
- Please check if you have udev rules installed: https://docs.nitrokey.com/nitrokey3/linux/firmware-update.html#troubleshooting
```

The only way to recover the key is to unplug it then plug it back again...
Note that I still can access stored TOTP secrets with `nitropy nk3 secrets get <secret name>`  (if I can remember how I called the secret entry in the key...) !

`nitropy` is deployed on my Ubuntu 23.04 using `pipx` and is up to date:
```shell
rico@ricolapscc:~$ nitropy version
Command line tool to interact with Nitrokey devices 0.4.38
0.4.38
```
And here are my nk3 details:
```shell
rico@ricolapscc:~$ nitropy nk3 status
Command line tool to interact with Nitrokey devices 0.4.38
UUID:               2A5AC5FE19D3345BB609DDEB7F3DF366
Firmware version:   v1.5.0
Init status:        ok
Free blocks (int):  33
Free blocks (ext):  463
Variant:            LPC55
```

Hope this helps

pfactum, 2023-06-22:
Mine one, NK3A NFC, is affected as well:

```
nitropy nk3 status
Command line tool to interact with Nitrokey devices 0.4.38
UUID:               2A0ED288B66C0F53AB17A33ED684B27A
Firmware version:   v1.5.0
Init status:        ok
Free blocks (int):  36
Free blocks (ext):  459
Variant:            LPC55

$ nitropy nk3 secrets list
Command line tool to interact with Nitrokey devices 0.4.38
Please provide PIN to show PIN-protected entries (if any), or press ENTER to skip
Please touch the device if it blinks
Current PIN (8 attempts left):
Please touch the device if it blinks
```

The device blinks, I touch it, LED turns red.

szszszsz, 2023-06-23:
Good news. I've found the problem, and application works stable with it so far.
Following looks like a good workaround until the new firmware will be released:
```
$ pipx run pynitrokey==v0.4.37 nk3 secrets list 
```

closed 2023-06-27 by szszszsz