Nitrokey/nitrokey-3-firmware #298: NK3A v.1.5.0 OpenPGP stops working if touch confirmation enabled (any UIF is set to "on")

jgoolde, 2023-06-21:
NK3A with firmware v.1.5.0
gpg v 2.3.8 on Linux

Assigned new pin, admin pin and reset code.
Generated 3 4096 bit RSA keys via  on the NK.
Set UIF to "on" for all actions.

gpg encrypt and decrypt worked for about 10-15 time.
After that, all pins were reset to default factory set values. Touch confirmation stopped working completely despite `gpg --card-status` showing all UIF settings on. Most importantly decryption stopped working as well , gpg reports `card error`.

I have tried resetting UIF to "off" but that did not solve the problem.
I can still see the same keys fingerprints but decryption does not work,

Another likely related issue: if any UIF settings is set to "on" before writing RSA keys to NK3, generating keys on the card or `keytocard` always fails with same "card error".

So far, I have performed factory reset and generated new keys without setting any UIF to "on", it seems to be working (so far).

sosthene-nitrokey, 2023-06-22:
Hello, thank you for the report.

> Generated 3 4096 bit RSA keys via gpg on the NK

Did you generate them or import them? It is normally not possible to generate RSA key larger than 2048 bit on-device.

Did it happen only once or is it something that you are able to  consistently if when you enable UIF?

What were the last operations you performed before the PINs were reset to their default?

jgoolde, 2023-06-22:
I followed instructions in https://docs.nitrokey.com/nitrokey3/linux/openpgp-keygen-on-device 
Once the key attributes are changed to 4096, it is exactly what NK3 shows.

The second time, I generated the keys outside of NK, following https://docs.nitrokey.com/nitrokey3/linux/openpgp-keygen-backup except for the key size where I chose 4096 instead of 2048

The issue with pin reset and card error happened both times.

> What were the last operations you performed before the PINs were reset to their default?
First time it was after I attempted to link NK3 with OpenKeychain via NFC - at that time I was not aware the NFC function does not work with OpenPGP.

Second time, I think it was `gpg --decrypt` - I was using that a lot (around 15 times).

sosthene-nitrokey, 2023-06-23:
>  Once the key attributes are changed to 4096, it is exactly what NK3 shows.

I do not understand what you mean in this sentence. Could you please clarify?

Besides the PIN, was other data affected? For example the metadata you might have configured with `gpg --card-edit` or the Fido and secrets app?

sosthene-nitrokey, 2023-06-23:
Hi,

It looks like the NFC usage, despite being unsupported can indeed cause the PINs to be reset due to an oversight in the implementation of the initialization step. This bug will be fixed in the next release of the firmware, see #301  and https://github.com/Nitrokey/opcard-rs/issues/165 for the relevant issues.

Meanwhile, we recommend you do not attempt to use OpenPGP over NFC.

closed 2023-07-04 by sosthene-nitrokey