Nitrokey/nitrokey-3-firmware #300: 0.4.38 cannot list secrets

orolhawion, 2023-06-22:
After I upgraded to `0.4.38` I did the following:

1. add password to password safe (3 entries overall)
2. tried to list secrets which did not work

(!) I do not know if 1 and 2 are related

```
$ nitropy nk3 secrets list
Command line tool to interact with Nitrokey devices 0.4.38
Please provide PIN to show PIN-protected entries (if any), or press ENTER to skip
Please touch the device if it blinks
Current PIN (8 attempts left):
Please touch the device if it blinks
Critical error:
An unhandled exception occurred
	Exception encountered: OSError('Failed reading a response')
Warning: no backend was found to use for communication. Please refer to documentation how to install additional libraries.
```
The NK3C then is red permanently and no more actions are possible for the following reason:

```
$ nitropy nk3 secrets list
Command line tool to interact with Nitrokey devices 0.4.38
Critical error:
An unhandled exception occurred
	Exception encountered: CtapError('CTAP error: 0x06 - CHANNEL_BUSY')
Warning: no backend was found to use for communication. Please refer to documentation how to install additional libraries.
```

szszszsz, 2023-06-23:
Good news. I've found the problem, and application works stable with the fix for it so far.
Following looks like a good workaround until the new firmware will be released:
```
$ pipx run pynitrokey==v0.4.37 nk3 secrets list 
```

orolhawion, 2023-06-23:
Nice, thanks for your effort to fix it. :)

closed 2023-06-27 by szszszsz