Nitrokey/nitrokey-3-firmware #310: Update from 1.4 to 1.5 ended with Exception('Wrong channel') during finalization

fredsleeve, 2023-07-07:
Used nitropy-v0.4.38.exe to update from 1.4 to 1.5 and ended up with "Wrong Channel" exception in step Finalize upgrade:
[cmdoutput.txt](https://github.com/Nitrokey/nitrokey-3-firmware/files/11984020/cmdoutput.txt)
[nitropy.log.mbmimil_.log](https://github.com/Nitrokey/nitrokey-3-firmware/files/11984027/nitropy.log.mbmimil_.log)

Nk3 test with pin looks good. Should I rerun the update or let it be?

`Command line tool to interact with Nitrokey devices 0.4.38
Found 1 Nitrokey 3 device(s):
- Nitrokey 3 at \\?\hid#vid_20a0&pid_42b2&mi_01#8&2301b96c&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}

Running tests for Nitrokey 3 at \\?\hid#vid_20a0&pid_42b2&mi_01#8&2301b96c&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
[1/4]   uuid            UUID query                      SUCCESS         734F6058513970990000000000000000
[2/4]   version         Firmware version query          SUCCESS         v1.5.0
[3/4]   status          Device status                   SUCCESS         Status(init_status=<InitStatus.0: 0>, ifs_blocks=239, efs_blocks=478, variant=<Variant.NRF52: 2>)
Please press the touch button on the device ...
Please press the touch button on the device ...
[4/4]   fido2           FIDO2                           SUCCESS

4 tests, 4 successful, 0 skipped, 0 failed

Summary: 1 device(s) tested, 1 successful, 0 failed`

robin-nitrokey, 2023-07-07:
Nothing to worry about.  Finalization basically means waiting for the device to boot up and ensuring that it reports the right version.  The wrong channel error could be caused by some other application (web browser?) trying to access it at the same time.  As long as it reports the correct version after the update, everything is fine.

fredsleeve, 2023-07-07:
Thanks for this quick reply. I just saw that "Nitrokey-App" was still open because I first wanted to update without cli. Maybe this caused it? Nevermind, I think this can be closed.

closed 2023-07-07 by robin-nitrokey