Nitrokey/nitrokey-3-firmware #313: Add stack analysis tool

sosthene-nitrokey, 2023-07-12:
With `-Z emit-stack-sizes` we can get the stack sizes of each function in the final firmware.

Steps:
1. use nightly
2. add to the linker script:

```

sosthene-nitrokey, 2023-07-12:
Other tools that can help make the output more readable: c++filt

sosthene-nitrokey, 2023-07-12:
Branch [stack-sizes](https://github.com/Nitrokey/nitrokey-3-firmware/tree/stack-sizes) contains the hacks used.

szszszsz, 2023-07-12:
I've improved output format, env call for `+x`, and it reads from stdin now.

```rust
#!/usr/bin/env -S cargo +nightly -Zscript

//! ```cargo
//! [package]
//! edition = "2021"
//! [dependencies]
//! stack-sizes = "0.5.0"
//!```

use std::io::Read;
use std::io;

use stack_sizes::analyze_executable;
fn main() {
    let mut stdin = io::stdin().lock();
    let mut buffer = Vec::new();
    stdin.read_to_end(&mut buffer).unwrap();
    let functions = analyze_executable(&buffer).unwrap();
    let mut sorted: Vec<_> = functions.defined.into_values().collect();
    sorted.sort_by_key(|f| f.stack().unwrap_or(0));
    for f in sorted {
        println!("{:5?} {:5} {:?}", f.stack(), f.size(), f.names() );
    }
}

```

robin-nitrokey, 2023-10-06:
Next iteration:
- reverse sort order (highest stack usage first)
- demangle identifiers
- make output prettier

```rust
#!/usr/bin/env -S cargo +nightly -Zscript

//! ```cargo
//! [package]
//! edition = "2021"
//! [dependencies]
//! stack-sizes = "0.5.0"
//! symbolic = { version = "12.4.1", features = ["demangle"] }
//!```

use std::io::Read;
use std::io;

use stack_sizes::analyze_executable;
use symbolic::demangle;

fn main() {
    let mut stdin = io::stdin().lock();
    let mut buffer = Vec::new();
    stdin.read_to_end(&mut buffer).unwrap();
    let functions = analyze_executable(&buffer).unwrap();
    let mut sorted: Vec<_> = functions.defined.into_values().collect();
    sorted.sort_by_key(|f| f.stack().unwrap_or(0));
    for f in sorted.into_iter().rev() {
        if let Some(stack) = f.stack() {
            print!("{:6}", stack);
        } else {
            print!("None");
        }
        print!(" {:6} ", f.size());
        for (i, name) in f.names().into_iter().enumerate() {
            if i > 0 {
                print!(", ");
            }
            print!("{}", demangle::demangle(name));
        }
        println!();
    }
}
```

Example output:
```
108568    652 lpc55_runner::app::rtic_ext::main::__rtic_init_resources
 22920    264 <iso7816::command::Command<_> as core::convert::TryFrom<&[u8]>>::try_from
 22896    176 usbd_ctaphid::pipe::Pipe<Bus>::handle_response
 22320  14032 ctap_types::ctap2::Request::deserialize
 22152   3508 embedded_runner_lib::soc::init::Stage2::next
 21800   6020 OS_EVENT
 21440    408 apdu_dispatch::dispatch::ApduDispatch::handle_reply
 19832  15260 trussed::service::ServiceResources<P>::reply_to
 18064   1048 webcrypt::ctap_app::try_handle_ctap2
 17392    364 webcrypt::ctap_app::try_handle_ctap1
```