Nitrokey/nitrokey-3-firmware #328: Fido compatiblity issues with Firefox >= 115

sosthene-nitrokey, 2023-07-25:
Registration of U2F seem to fail with Firefox 115 and 116 (developer edition) on Codeberg and Discourse forums.

It appears to work on chromium and Firefox 102.13.0esr.

Not tested with 114, but It might have been introduced in a recent update.

Once registered however using U2F for login appears to work properly.

https://support.nitrokey.com/t/nitrokey3-fido2-does-not-work-with-forgejo

sosthene-nitrokey, 2023-07-25:
More info:

It fails whether or not the pin is set.
Logs from a dev device show a CTAP2 Error 12, which corresponds to `InvalidCbor` (from ctap_types).

robin-nitrokey, 2023-07-26:
My test with Firefox 115 and a dev device on support.nitrokey.com also caused an `InvalidCbor` error.  The CBOR data was:

```
01 A6 01 58 20 42 52 25 C5 21 06 E9 2E 67 EA 11 BE 77 CE 4D 0A 92 6E CA BB A9 97 ED 56 D6 05 3A 7A 85 1D B3 19 02 A1 62 69 64 74 73 75 70 70 6F 72 74 2E 6E 69 74 72 6F 6B 65 79 2E 63 6F 6D 03 A2 62 69 64 58 28 61 61 34 38 39 32 64 65 32 33 37 63 33 62 30 34 36 35 62 65 63 66 38 36 31 31 38 62 65 62 31 38 30 66 65 66 66 61 62 34 64 6E 61 6D 65 6E 72 6F 62 69 6E 2D 6E 69 74 72 6F 6B 65 79 04 8D A2 63 61 6C 67 26 64 74 79 70 65 6A 70 75 62 6C 69 63 2D 6B 65 79 A2 63 61 6C 67 38 22 64 74 79 70 65 6A 70 75 62 6C 69 63 2D 6B 65 79 A2 63 61 6C 67 38 23 64 74 79 70 65 6A 70 75 62 6C 69 63 2D 6B 65 79 A2 63 61 6C 67 38 2E 64 74 79 70 65 6A 70 75 62 6C 69 63 2D 6B 65 79 A2 63 61 6C 67 27 64 74 79 70 65 6A 70 75 62 6C 69 63 2D 6B 65 79 A2 63 61 6C 67 38 24 64 74 79 70 65 6A 70 75 62 6C 69 63 2D 6B 65 79 A2 63 61 6C 67 38 25 64 74 79 70 65 6A 70 75 62 6C 69 63 2D 6B 65 79 A2 63 61 6C 67 38 26 64 74 79 70 65 6A 70 75 62 6C 69 63 2D 6B 65 79 A2 63 61 6C 67 04 64 74 79 70 65 6A 70 75 62 6C 69 63 2D 6B 65 79 A2 63 61 6C 67 05 64 74 79 70 65 6A 70 75 62 6C 69 63 2D 6B 65 79 A2 63 61 6C 67 06 64 74 79 70 65 6A 70 75 62 6C 69 63 2D 6B 65 79 A2 63 61 6C 67 07 64 74 79 70 65 6A 70 75 62 6C 69 63 2D 6B 65 79 A2 63 61 6C 67 39 01 00 64 74 79 70 65 6A 70 75 62 6C 69 63 2D 6B 65 79 05 83 A2 62 69 64 58 B7 A3 00 58 93 C5 A5 27 B0 16 1A 91 49 9D CA B2 B7 62 9F B0 F4 A6 DC A7 A6 A5 C3 4B 08 0D 4D 59 4B A4 C4 BA 1F 5F 65 28 66 3E 1F 15 43 79 63 ED 58 76 A5 C5 CE 6D 1F 59 FB 66 F7 4A 22 73 8D 37 0A 54 03 9C E5 A7 1B 31 1D 75 6D 3D A3 DD 15 FE 60 FE 24 72 0C 8A 76 D3 78 43 6C 9E 69 4D 69 65 95 9D 02 F7 F1 D7 1C 94 BA 86 0D C5 BC A9 DA 02 87 41 08 31 B3 C0 AF D1 DB AA 26 65 69 6A D0 C3 16 6C 9C 9F 19 F0 9C FA D3 AD B6 50 B6 24 7A F7 5C CD 63 5E 55 21 59 DA 01 4C 94 E0 C1 F9 60 E8 97 6F 92 74 CD AB 02 50 A6 E6 E9 36 9B 01 E1 32 60 16 BB B1 A6 67 4F 70 64 74 79 70 65 6A 70 75 62 6C 69 63 2D 6B 65 79 A2 62 69 64 58 B7 A3 00 58 93 ED 0C 91 F5 8B D2 A6 F0 19 BB 31 82 60 31 D5 D8 61 04 6F 40 B8 5F 29 F5 A8 BD A2 9C 1E A5 72 BC 98 FC 0B 6E 58 EB AE B1 C7 4D FF AE F8 03 09 D8 80 64 FB 6B E3 59 71 68 36 45 6F 6F 4E 62 CF 4D 1A D8 D9 2D 80 A3 20 30 7B 83 D9 D4 42 4F 46 74 9A 78 A0 8F 07 9A 02 41 60 BA D2 65 F3 89 8B F2 70 C5 C0 B5 8B C7 FD 8A 92 71 19 05 D3 78 9A FE 7A DF 69 BF 76 63 B3 59 7F 91 7F 04 E3 81 FF C9 9C D9 BA 7B 4E 21 F5 92 74 44 A5 5C 7A AB C1 90 85 88 0B 01 4C 2A 6E 13 75 28 1E 6C 87 CD D1 6E DE 02 50 C7 BC F3 B7 5E 4C 8B CA D7 2D C5 F2 EB 69 F0 CF 64 74 79 70 65 6A 70 75 62 6C 69 63 2D 6B 65 79 A2 62 69 64 58 B6 A3 00 58 92 5B 28 91 3F 14 37 4C 2A ED 88 94 B6 D2 AD 4F 3F 66 57 D3 63 74 B2 5F 57 7D 3B 8A 39 3E B7 DA 22 17 43 4E 4E 99 BA 5B 06 53 DA 90 E9 A9 3C 27 1D 11 DF 69 77 33 78 F0 F2 EA BE 88 B4 18 A8 1F B0 8B 5B 03 30 2D 04 48 D7 7E 83 A2 C3 6F AC DF B8 0F 9B BF 5E F8 BB 42 F1 C3 3E 5A E2 F7 B4 AD EF D7 E2 7B BE 08 02 70 2C FF F0 1A C9 25 2E DF 24 E7 4D 10 91 FF 2C 15 98 B7 AB F7 79 7F 08 4E 27 9D 4E 2A BC B4 1B 39 AF 36 38 F5 C6 84 B5 D3 E2 D4 F9 01 4C AA 25 88 B2 10 E7 7F C2 CA E7 96 57 02 50 EA F7 75 AE 09 9B F0 41 64 B1 4C 60 D2 F0 14 38 64 74 79 70 65 6A 70 75 62 6C 69 63 2D 6B 65 79 07 A1 62 72 6B F4
```

`serde_path_to_error` points to a `SerdeDeCustom` error in field `04` which would be `pub pub_key_cred_params: Vec<PublicKeyCredentialParameters, 12>` (see [Request](https://docs.rs/ctap-types/latest/ctap_types/ctap2/make_credential/struct.Request.html#)).  Unfortunately, `cbor-smol`’s errors are not very instructive.

Inspecting the data with `ciborium` shows that the array in question contains 13 elements while we only support 12.  If I bump the capacity, parsing works fine.  Now the question is what we should use instead.

The data structure could be optimized by removing `key_type` from `PublicKeyCredentialParameters`.  Currently, the spec says that this field should always be `"public-key"` and entries with other values should be ignored.  If we would enforce this during deserialization instead of request handling, we could save lots of space and easily bump the capacity of the vec to a much higher number.

cc @msirringhaus

msirringhaus, 2023-07-27:
I can confirm your find. When modifying the register-example of authenticator-rs to contain 13 key_types, I do indeed also get an InvalidCbor-error from NK3, while other tokens work fine.

Stating the obvious, but just for completeness: From the browser-side, we can't *not* send `key_type`, as it's required.

Not sure about your internals, but maybe you could also deserialize this string to an enum (with currently only one possible value) to reduce space?

PR #331 (2023-09-05): Use updated ctap-types and fido-authenticator for better compatibility with Firefox
Partially fix #328 . Registration still fails with discourse but now it gets to a internal server error instead of a failed CBOR deserialization.
closed 2023-09-05 by sosthene-nitrokey