Nitrokey/nitrokey-3-firmware #330: GPG keytocard fails with card inaccessible afterwards

Unb0rn, 2023-08-01:
Met a strange behavior - tried to replace auth key on my NK3 and while doing so I got an error with a message: `gpg: error getting current key info: General error`

Nitrokey 3 FW: v1.5.0-test.20230704 (wanted to play with PIV a bit)
GPG: 2.3.8

Steps to reproduce:
- Add an ECC key (NISTp521, Auth) with addkey
- Select it
- Run keytocard

The card already has a key in auth slot (ed25519) and I want to replace it.

After this error any communication with the card fails with something like:
```
gpg --card-status       
gpg: OpenPGP card not available: Not supported
```
Unplugging and replugging the card ends up with:
```
gpg --card-status
gpg: selecting card failed: No such device
gpg: OpenPGP card not available: No such device
```
Killing and restarting scdaemon seems to help

Unb0rn, 2023-08-02:
Tried updating GPG to 2.4, "downgrading" NK3 firmware to 1.5.0, re-generating the key to NISTp256 - it only seemed to set key attributes to nistp256 but the key is not uploaded and the error remains the same - `gpg: KEYTOCARD failed: Card error`

sosthene-nitrokey, 2023-08-02:
The GPG function of the Nitrokey 3 doesn't support NIST P521yet.

If you want this functionality to be added, please voice your support in https://github.com/Nitrokey/opcard-rs/issues/36 so that we prioritize this feature.

> Tried updating GPG to 2.4, "downgrading" NK3 firmware to 1.5.0, re-generating the key to NISTp256 - it only seemed to set key attributes to nistp256 but the key is not uploaded and the error remains the same - gpg: KEYTOCARD failed: Card error

I am not sure I fully understand this comment. Are you locked out of generating/importing a nistp256 key? This would be a bug.

Unb0rn, 2023-08-02:
Initially I generated NISTp521 but today I tried several things - updated GnuPG, downgraded Nitrokey firmware to a stable version and finally generated NISTp256.

When I tried putting this new NISTp256 key to card (via GnuPG 2.4 this time if it matters) - it failed with Card error but changed authentication key attrbutes (the one I tried to push to card) from ed25519 to nistp256 and now this key on card is completely unusable and I'm afraid I have to reset my NK3 completely.

Regarding the p521 part - it's a bit off-topic - but I need it unless Nitrokey is interested in fixing OpenKeychain for Android - opened an [issue](https://github.com/open-keychain/open-keychain/issues/2769) in this project. In a nutshell - it works with some key lengths and curves but fails or even crashes with the others. So, I voted on that opcard-rs issue.

sosthene-nitrokey, 2023-08-02:
Changing the key attribute will delete the key stored on the NK3, so it is to be expected that you Ed25519 key cannot be used anymore.

I am not able to reproduce the error you encountered. Importing a nistp256 with an Ed25519 key in the authentication key works properly.

Are the other slots of your keys also populated?

closed 2023-09-28 by sosthene-nitrokey