Nitrokey/nitrokey-3-firmware #347: fido: unable to authenticate to gandi.net using firefox

jplejacq-quoininc-com, 2023-09-25:
firmware-version: v1.5.0-test.20230704
firefox-version: 117.0.1 (64-bit)
os: debian 11.7 (Bullseye)

After registering my NK3 using chrome, I'm unable to authenticate using the NK3 using firefox. Instead, the login dialog hangs and the LED on the NK3 goes red. Chrome works fine with the NK3.

I should note that gandi attempts to gather more details information from the NK3 and Firefox pops up a warning dialog. Chrome does not do this.

You may want to track this in the related issue: TRACKING: Incompatibility issues #269

daringer, 2023-09-25:
thanks for the report, could you try to disable `ctap2` within firefox and try again?
You should find it in `about:config` search for `security.webauthn.ctap2` and switch it to `false`. CTAP2 support is pretty new in firefox, maybe something is wrong there.
On the other side the red LED indicates that something is weird with the communication. 
Does register work with firefox?

robin-nitrokey, 2023-09-26:
This looks similar to issue https://github.com/Nitrokey/nitrokey-3-firmware/issues/328 that will be fixed in the next firmware release.

jplejacq-quoininc-com, 2023-09-26:
Thanks you for the quick response.

1. Registration did not work with firefox.
2. I get the same results with ctap2 disabled in firefox.

daringer, 2023-09-26:
Ok, thanks, then most likely this should be solve by #328 as @robin-nitrokey mentioned. We expect a new test-release soon(tm), which will include this fix, would be great if you could re-test then.

robin-nitrokey, 2024-03-28:
@jplejacq-quoininc-com This should be fixed in test release [v1.5.0-test.20231026](https://github.com/Nitrokey/nitrokey-3-firmware/releases/tag/v1.5.0-test.20231030) and stable release [v1.6.0](https://github.com/Nitrokey/nitrokey-3-firmware/releases/tag/v1.6.0).  Please re-open if this problem still occurs.

closed 2024-03-28 by robin-nitrokey