Nitrokey/nitrokey-3-firmware #348: openpgp: gpg-agent fails for ed25519 keys with ssh certificates

jplejacq-quoininc-com, 2023-09-25:
It's not clear if this is a firmware issue or the openssh protocol itself.

firmware: v1.5.0-test.20230704
gnupg: 2.4.3
openssh: OpenSSH_for_Windows_9.2p1, LibreSSL 3.7.2
os: Windows 11 Pro, fully updated

Here is the scenario:

* Create a ed25119 authentication subkey and transfer to the nk3 device using gpg.
* Create a ssh certificate for this subkey using default ssh-keygen options.
* Try to connect to server using certificate. We get the following error:

sign_and_send_pubkey: signing failed for ED25519 "~/.ssh/ed25519.pub": agent refused operation

NIIBE Yutaka, author of Gnuk and Nitrokey Start, has a good explanation of the root cause of the problem and a possible workaround [1].

It would be awesome if the NK3 firmware could expand the memory to handle the new ssh protocol features.

[1] https://dev.gnupg.org/T6250

sosthene-nitrokey, 2023-09-26:
Hi, thank you for the report. 

In the GnuPG ticket you mention that the signed data is around 700 bytes, but are you sure that it's the Nitrokey 3 that rejects such large data? After some manual checking with Ed25519 it can sign up to 1024 bytes. (actually I also noticed that it crashes for 1025 https://github.com/Nitrokey/opcard-rs/issues/173 :worried: ).

So if it's only 700 bytes as you say I don't think the error comes directly from the Nitrokey 3 (and if it did you would have noticed a crashing key). I think we could bump it to a bit less than 2KB but that doesn't seem like the proper way to fix this problem.

jplejacq-quoininc-com, 2023-09-26:
Thank you for the quick response.

So far I have had no issue signing ed25519 keys with either gpg or openssh.

I agree with you that this solution is not optimal. There is definitely a protocol issue here as NIIBE mentions. However, it would make NK3 a bit more robust to poorly specified protocol.

sosthene-nitrokey, 2023-09-26:
I will leave this open for documentation. If anyone encounters this issue, the recommended solutions are to either use another protocol that can do pre-hashing (P256 or RSA), or to use the OpenSSH integration with FIDO if you are using a sufficiently recent version of OpenSSH.

hardfalcon, 2026-11-05:
I think the above-mentioned limitation that the Nitrokey 3 won't sign more than 1024 bytes is also causing login failures on SSH hosts that use an `ssh-mldsa44-ed25519` host key (which is enabled by default since OpenSSH 10.6).

This is because ssh-mldsa44-ed25519 have a size of [1344 bytes](https://datatracker.ietf.org/doc/html/draft-miller-sshm-mldsa44-ed25519-composite-sigs-00#name-composite-raw-public-key).

One possible workaround is to configure the SSH client to only use the hosts `ssh-ed25519` host key.

Another (IMO worse) workaround is to turn off the SSH protocol extension that includes the host's public key in the data signed by the client key to make sure that signature can only be used to log into that specific SSH host. With OpenSSH, that feature can be disabled using `ssh -o PubkeyAuthentication=unbound someuser@somehost`.

Are those 1024 bytes a hardware limitation, or would it be possible to increase that number by a few hundred bytes to make the Nitrokey 3 work with PQC-enabled SSH hosts?

I want to thank @blacklotus for helping me debug this, making me aware of the Nitrokey's 1024 byte limit in general and this Github issue in particular, and for proposing to try OpenSSH's `PubkeyAuthentication=unbound` option as a possible workaround if the host doesn't offer any host keys that are "small enough" to fit the Nitrokey's 1024 byte limit.