Nitrokey/nitrokey-3-firmware #389: When will PIV functionality be stable?

anotherbridge, 2023-11-24:
I would be interested if you already have an estimate on when the PIV functionality will be considered as stable?

Thanks a lot in advance!

jplejacq-quoininc-com, 2023-12-04:
>     1. Can you share any high-level use-case you would like to use PIV for?

At a company level we have three use cases, list in priority:

1. VPN (IPSec based) using self managed certificate authority.
2. System authentication primarily for macOS and Linux, secondarily Windows AD.
3. Possibly S/MIME.

>     2. Does your use-case need a r/w-enabled Mini-Driver for Windows?

No.

>     3. There is a generic ro-enabled Mini-Driver within OpenSC, is this enough?

Yes.

>     4. Which of the many PIV extensions are preferable for you?

The only one at the moment is curve 25519 key support. This is something we are exploring.

>     5. How would you provision PIV on the Nitrokey 3? Do you have your own (FOSS?) tooling? Or would you need us to provide this tooling, too?

FOSS tooling based on Linux would be the preferred solution.

SunStrom, 2023-12-23:
Our use case is basically identical, just with slightly different priorities:

>     1. Can you share any high-level use-case you would like to use PIV for?

1. System authentication (Windows)
2. VPN auth (OpenVPN)
3. possibly S/MIME

>     2. Does your use-case need a r/w-enabled Mini-Driver for Windows?

No.

>     3. There is a generic ro-enabled Mini-Driver within OpenSC, is this enough?

Yes.
 
>     4. Which of the many PIV extensions are preferable for you?

ECC, curve 25519 if possible. 
 
>     5. How would you provision PIV on the Nitrokey 3? Do you have your own (FOSS?) tooling? Or would you need us to provide this tooling, too?

As far as I understand, provisioning is already possible with FOSS tooling (just neither easy nor comfortable) if you have the needed master key for the card - this would be enough.

tgahlx, 2024-04-11:
I would also like to vote for this feature as we want to rollout tokens in our enterprise and would rather not buy yubikeys.
We like this idea to have a single smartcard that supports multiple use cases like piv and passkeys and others together.

> 1. Can you share any high-level use-case you would like to use PIV for?

Cert based Authentication and Authorization on different Systems and services and S/MIME.

> 2. Does your use-case need a r/w-enabled Mini-Driver for Windows?

Not now.

> 3. There is a generic ro-enabled Mini-Driver within OpenSC, is this enough?

Yes

> 4. Which of the many PIV extensions are preferable for you?

Support Keys as stated in the FAQ
https://docs.nitrokey.com/nitrokey3/faq

> 5. How would you provision PIV on the Nitrokey 3? Do you have your own (FOSS?) tooling? Or would you need us to provide this tooling, too?

using nitropy, piv-tool or default cli tools like pkcs11/15

hailfinger, 2025-03-21:
This might be partially or completely done with the release of firmware version 1.8.1

jans23, 2025-03-21:
Current [firmware 1.8.0](https://www.nitrokey.com/news/2025/nitrokey-3-firmware-18-piv-windows-login-nist-p-521-brainpool) released in January contains PIV already which we consider stable.

closed 2025-03-21 by jans23