Nitrokey/nitrokey-3-firmware #391: Error while Yubikey NEO is connected

jwgn, 2023-11-24:
Hi,

I use a Yubikey NEO along with my NK3 (for static password and some sites which only support [Yubico OTP](https://developers.yubico.com/OTP/)).

As long as the Yubikey is connected to my PC, I am unable to interact with my NK3:

```
PS Z:\> nitropy.exe fido2 list-credentials
Command line tool to interact with Nitrokey devices 0.4.4
Please provide pin:
Critical error:
An unhandled exception occurred
Exception encountered: NonUniqueDeviceError()
```

I am also unable to register the NK3 for U2F on any website.

When I remove the Yubikey, everythink works fine. Happened on `v1.5.0`, just upgraded to `v1.6.0-rc.1`, same issue.

Is this expected behaviour? The Yubikey NEO doesn't even support FIDO.

daringer, 2023-11-30:
The first issue: accessing the Nitrokey 3 while another FIDO2 device is present should be solvable by also passing the specific device like that:
```
nitropy fido2 list-credentials -s device=/dev/hidraw11
```
With the device being what is written in the first column in `nitropy nk3 list`. At least this is what works (for Linux) if you have connected more than one Nitrokey 3.

The other issue, the U2F register is pretty weird, also just tested this successfully on webauthn.io using 2x Nitrokey 3. Registering the device works with both devices and both will ask for the user confirmation (user presence) and the one which is confirmed will then be registered for the service.

On a first glance this looks like the Yubikey Neo does not behave as expected (defined by the spec). Although it has no FIDO2 capabilities it comes with U2F (which likely is also the reason the *non-unique* error is shown by `nitropy`) and my guess would be that this is roughly the same reason why registering won't work if both devices are plugged in.

jwgn, 2023-12-01:
Thanks for your answer. Do you have any idea how to select the device on Windows?

```
nitropy.exe nk3 list
Command line tool to interact with Nitrokey devices 0.4.42
:: 'Nitrokey 3' key\\?\hid#vid_20a0&pid_42b2&mi_01#c&22afc97d&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}: Nitrokey 3 D5DAB95933C43C5691135B4681EC888E
```

Tried various parts from that output but it always throws errors like `ArgumentError("argument 1: <class 'TypeError'>: wrong type")`.