Nitrokey/nitrokey-3-firmware #393: Finalize update v1.6.0 with NK3 USB-A critical error

hase77, 2023-11-30:
Just upgraded my both NK3 with firmware 1.6.0. USB-C give no errors, but USB-A does.

```
Command line tool to interact with Nitrokey devices 0.4.43
Do you want to download the firmware version v1.6.0? [Y/n]: y
Download v1.6.0: 100%|█████████████████████████████████████████████████████████████| 909k/909k [00:00<00:00, 8.27MB/s]
Current firmware version:  v1.5.0
Updated firmware version:  v1.6.0

Please do not remove the Nitrokey 3 or insert any other Nitrokey 3 devices during the update. Doing so may damage the Nitrokey 3.
Do you want to perform the firmware update now? [y/N]: y

Please press the touch button to reboot the device into bootloader mode ...

Perform firmware update: 100%|█████████████████████████████████████████████████████| 461k/461k [00:08<00:00, 52.2kB/s]
Finalize upgrade:  98%|█████████████████████████████████████████████████████████████▋ | 98/100 [00:46<00:00,  2.12%/s]
Critical error:
No Nitrokey 3 device found

--------------------------------------------------------------------------------
Critical error occurred, exiting now
Unexpected? Is this a bug? Would you like to get support/help?
- You can report issues at: https://support.nitrokey.com/
- Writing an e-mail to support@nitrokey.com is also possible
- Please attach the log: 'C:\Users\ms\AppData\Local\Temp\nitropy.log.pw4yj_cw' with any support/help request!
--------------------------------------------------------------------------------
```

Stick is not recognized anymore!
[nitropy.log.3xdh7st8.txt](https://github.com/Nitrokey/nitrokey-3-firmware/files/13513562/nitropy.log.3xdh7st8.txt)

robin-nitrokey, 2023-11-30:
Thank you for the report.  We will look into the issue.  Looking at the log, I don’t see anything suspicious.  Let me know if you want to try to debug this on your device.  This would require you to open the casing of the device.

hase77, 2023-11-30:
Even if I would like to help further, I probably lack the appropriate tools and knowledge.

robin-nitrokey, 2023-11-30:
@hase77 It is not too complicated – once you have opened the casing, you only have to connect two pads with a cable or some other conducting material as depicted here: https://user-images.githubusercontent.com/12674969/203349391-3feac7f8-5127-40df-b66c-910c1a41f975.jpg  If you connect the device to your computer while the pads are connected, it will boot into bootloader mode and you can use `nitropy` to install a different firmware version and see if that fixes the problem.

hase77, 2023-12-01:
Ok, thanks. But sadly the same error after the finalize:
[nitropy.log.5it6hsfy.txt](https://github.com/Nitrokey/nitrokey-3-firmware/files/13523466/nitropy.log.5it6hsfy.txt)

```
PS C:\Users\ms\Desktop\Stick\Fido2> .\nitropy-v0.4.43-x64-windows-binary.exe nk3 update --version v1.6.0-rc.1
Command line tool to interact with Nitrokey devices 0.4.43
Do you want to download the firmware version v1.6.0-rc.1? [Y/n]: y
Download v1.6.0-rc.1: 100%|████████████████████████████████████████████████████████| 897k/897k [00:00<00:00, 3.29MB/s]
Current firmware version:  [unknown]
Updated firmware version:  v1.6.0-rc.1

Please do not remove the Nitrokey 3 or insert any other Nitrokey 3 devices during the update. Doing so may damage the Nitrokey 3.
Do you want to perform the firmware update now? [y/N]: y
Perform firmware update: 100%|█████████████████████████████████████████████████████| 455k/455k [00:08<00:00, 52.1kB/s]
Finalize upgrade:  98%|█████████████████████████████████████████████████████████████▋ | 98/100 [00:37<00:00,  2.62%/s]
Critical error:
No Nitrokey 3 device found

--------------------------------------------------------------------------------
Critical error occurred, exiting now
Unexpected? Is this a bug? Would you like to get support/help?
- You can report issues at: https://support.nitrokey.com/
- Writing an e-mail to support@nitrokey.com is also possible
- Please attach the log: 'C:\Users\ms\AppData\Local\Temp\nitropy.log.5it6hsfy' with any support/help request!
```

robin-nitrokey, 2023-12-01:
Thank you for testing the release candidate.  We plan to publish a test release shortly that will improve diagnostics for one potential cause of this problem.  It would be great if you could give that one a try too.  I will let you know once it is available.

hase77, 2023-12-06:
Before flashing, the device is only found in boot mode:
```
PS C:\Users\ms\Desktop\Stick\Fido2> .\nitropy-v0.4.43-x64-windows-binary.exe nk3 list
Command line tool to interact with Nitrokey devices 0.4.43
:: 'Nitrokey 3' keys
\\?\hid#vid_20a0&pid_42dd#7&dc98c3c&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}: Nitrokey 3 Bootloader (LPC55) 8FE8BFC45160785B9CFE9B31F837F4FF
```

Flashing work, but resulted in red led:
```
PS C:\Users\ms\Desktop\Stick\Fido2> .\nitropy-v0.4.43-x64-windows-binary.exe nk3 update --version v1.6.0-test.20231206
Command line tool to interact with Nitrokey devices 0.4.43
Do you want to download the firmware version v1.6.0-test.20231206? [Y/n]: y
Download v1.6.0-test.20231206: 100%|█████████████████████████████████████████████| 1.20M/1.20M [00:00<00:00, 2.19MB/s]
Current firmware version:  v1.6.0-rc.1
Updated firmware version:  v1.6.0-test.20231206

Please do not remove the Nitrokey 3 or insert any other Nitrokey 3 devices during the update. Doing so may damage the Nitrokey 3.
Do you want to perform the firmware update now? [y/N]: y

Please press the touch button to reboot the device into bootloader mode ...

Perform firmware update: 100%|█████████████████████████████████████████████████████| 546k/546k [00:10<00:00, 52.4kB/s]
Finalize upgrade:  98%|█████████████████████████████████████████████████████████████▋ | 98/100 [00:32<00:00,  2.98%/s]
Critical error:
No Nitrokey 3 device found

--------------------------------------------------------------------------------
Critical error occurred, exiting now
Unexpected? Is this a bug? Would you like to get support/help?
- You can report issues at: https://support.nitrokey.com/
- Writing an e-mail to support@nitrokey.com is also possible
- Please attach the log: 'C:\Users\ms\AppData\Local\Temp\nitropy.log.kru_a872' with any support/help request!
```
After flashing and reboot the device is not found.

After some time I somehow managed to find the device for one time only.
```
PS C:\Users\ms\Desktop\Stick\Fido2> .\nitropy-v0.4.43-x64-windows-binary.exe nk3 list
Command line tool to interact with Nitrokey devices 0.4.43
:: 'Nitrokey 3' keys
\\?\hid#vid_20a0&pid_42b2&mi_01#8&21a26cca&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}: Nitrokey 3 8FE8BFC45160785B9CFE9B31F837F4FF

PS C:\Users\ms\Desktop\Stick\Fido2> .\nitropy-v0.4.43-x64-windows-binary.exe nk3 status
Command line tool to interact with Nitrokey devices 0.4.43
UUID:               8FE8BFC45160785B9CFE9B31F837F4FF
Firmware version:   v1.6.0-test.20231206
Init status:        ok
Free blocks (int):  41
Free blocks (ext):  478
Variant:            LPC55
```

It seems that 1 of several tries work to boot the stick. Sometime 3-4 tries, sometimes 20-30 :( 

[nitropy.log.kru_a872.txt](https://github.com/Nitrokey/nitrokey-3-firmware/files/13588460/nitropy.log.kru_a872.txt)

hase77, 2023-12-06:
If it boots correctly,:
```
Command line tool to interact with Nitrokey devices 0.4.43
Found 1 Nitrokey 3 device(s):
- Nitrokey 3 at \\?\hid#vid_20a0&pid_42b2&mi_01#8&21a26cca&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}

Running tests for Nitrokey 3 at \\?\hid#vid_20a0&pid_42b2&mi_01#8&21a26cca&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}

[1/5]   uuid            UUID query                      SUCCESS         8FE8BFC45160785B9CFE9B31F837F4FF
[2/5]   version         Firmware version query          SUCCESS         v1.6.0-test.20231206
[3/5]   status          Device status                   SUCCESS         Status(init_status=<InitStatus.0: 0>, ifs_blocks=33, efs_blocks=472, variant=<Variant.LPC55: 1>)
[4/5]   se050           SE050                           SKIPPED
Please press the touch button on the device ...
Please press the touch button on the device ...
[5/5]   fido2           FIDO2                           SUCCESS
```

daringer, 2023-12-06:
Okay okay, given this non-determinism this very much like a hardware issue - during the init-process inside the hardware there is not anything non-deterministic which might end up in such a situation.

Please write with your SOxxxxx (order number) to support (at) nitrokey (dot) com - so we can replace your faulty device. 
Thanks a lot for investigating this with us. 
Please close this issue, if the way forward is fine for you...

closed 2023-12-06 by hase77

hase77, 2023-12-10:
The new key arrived friday. Thanks for the repacement! I have just flashed the stick successfully to 1.6.0. Sorry for the circumstances, I use this stick as of my backups. Therefore I was not aware that it have an hardware issue! I never noticed this behavior before.