Nitrokey/nitrokey-3-firmware #398: Nitrokey 3C NFC FIDO authentication failure

jplejacq-quoininc-com, 2023-12-02:
Security key (FIDO) authentication fails (always fails) on when using NFC interface. Authentication works fine with USB C interface.

Hardware: Nitrokey 3C
Firmware: v1.5.0-test.20230704
Device: Pixel 6 Android 14
Browser: Firefox, Chrome, Brave

Steps:

1. Go to google.com and select login.
2. Provide user and password. Select Next. This is always successful.
3. Prompted for security key. This step is unreliable and often takes quite some time for device to be recognized. From the Nitrokey forums, this seems to be a know issue.
4. When device is finally recognized, site prompts to continue to hold Nitrokey next to device.
5. Screen displays "There was a problem. Try using your security key again or try another way to verify it's you." This always occurs.

A YubiKey 5C NFC works fine using the NFC interface. It is also immediately recognized by my phone.

jplejacq-quoininc-com, 2023-12-02:
I cannot edit the initial report so some more details here.

1. This failures happens on all sites I've tried.
4. When device is recognized, screen displays a) continue to hole security key next to device; b) you're all set, you can remove the security device c) authentication failure. try again or another way.

anotherbridge, 2024-09-11:
Is there any update on this? 

I am having the same issue across different devices. I tested 
* iPhone (different generations) 
* LineageOS (Fairphone 3) 
* Other Android phones

The firmware versions tested were: 
* v1.7.0
* v.1.7.2
* v1.7.2-test.20240813

Browsers: 
* Firefox
* Safari
* Chrome
* Brave

The phones can usually read out the data of the NFC tag, but FIDO authentication using the NFC interface won't work.

admirito, 2026-10-06:
Same symptom here, on the newer hardware revision, and it depends on the phone.

**Key:** Nitrokey 3C NFC, LPC55, hardware revision R2 (FM11NT082C; admin status reports revision 2), firmware v1.9.0, delivered October 2026.
  
**Fails over NFC:** Pixel 6 Pro and Pixel 7 Pro, Chrome, webauthn.io. "Continue to hold your security key", a check mark for a moment, then "hold" again, no PIN prompt, then "An unknown error occurred while talking to the credential manager." Fails for registration and authentication, discoverable and non-discoverable credentials, user verification discouraged or preferred, with and without a FIDO2 PIN set.

**Works over NFC:** same key, Pixel 9 Pro Fold: registration and authentication of a discoverable credential (with the PIN prompt), and authentication of a non-discoverable credential registered over USB.

**Works over USB-C** on the Pixel 6 Pro.

sometreee, 2026-11-07:
I'm having the same issues, newer pixel 9 works but my pixel 6 shows "lost contact with the security key"  in authnkey