Nitrokey/nitrokey-3-firmware #404: update v1.6.0 with NK3 USB-A failed

Ingo-Albrecht, 2023-12-11:
I saw you released v1.6.0 again, but unfortunately my update from 1.5.0 failed first try. Unlike https://github.com/Nitrokey/nitrokey-3-firmware/issues/393 it broke midway through the flash and now status does not recognise it, nitropy nk3 list shows it in bootloader mode.

How to proceed?

[image]
[nitropy.log.02sxeyo4.txt](https://github.com/Nitrokey/nitrokey-3-firmware/files/13629842/nitropy.log.02sxeyo4.txt)

sosthene-nitrokey, 2023-12-11:
Hi,
Thank you for the report. 

It looks like the firmware update failed, whereas in #393 the update succeed but then the device failed. Since the device is now stuck in bootloader mode, you should be able to update it again. Make sure that the device is correctly plugged-in to prevent any disconnection from making the update fail.

Ingo-Albrecht, 2023-12-11:
Hi,
yes, it worked second try, great.  I used the offline `--update firmware-nk3-v1.6.0.zip`.

FWIW: Unlike #393 the LED did stay off, also when I replugged it in bootloader mode (that darkness made me open the issue; this was the first nk3 update here).
After the successful update `nitropy nk3 status` did produce an error, also after `reboot`, until I re-plugged it again. 
Now it behaves normal and intact (fido2, a bunch of secrets, gpg4096).

I close this.
Cheers.

closed 2023-12-11 by Ingo-Albrecht