Nitrokey/nitrokey-3-firmware #421: Android: Unrecognized in Termbot and OpenKeyChain

ghagl, 2023-12-23:
nk3 firmware version: v1.6.0-test.20231218

Termbot reports "Internal error: Security Key returned error UNKNOWN (0xa3f)"

OpenKeyChain reports "Error: Failed to get pw status bytes".

I'm unsure whether the issue is NK3-related or just the apps themselves, but my old Nitrokey Start works without issues there.

sosthene-nitrokey, 2024-01-02:
Hi!

Are you using NFC or usb? OpenPGP smartcard functionality is not available over NFC.

ghagl, 2024-01-02:
Both failed, however NFC is of course moot if it's not intended to work.

levinuss, 2024-01-02:
Can confirm, I get the same error when trying to connect my NK3C NFC (Firmware `v1.6.0-test.20231218` with SE050 backend enabled) to OpenKeychain while it works fine with on my laptop.

daringer, 2024-01-02:
the issue is that there is a PR pending/in-development at OpenKeychain to make NK3 work in general, but the app seems to be in maintainance mode, so I do not have too many hopes that this will be pushed forward: https://github.com/open-keychain/open-keychain/pull/2842

ghagl, 2024-02-11:
There's a new version here https://github.com/open-keychain/open-keychain/commit/f099b86950b2b7a9449775e297b3f34a245929e2

Expected to work with NK-3. Waiting to test this as this is not yet packaged on F-droid

daringer, 2024-02-11:
jup jup, we have managed to arrange something with the guys at open-keychain, once this is fully working we sure will communicate this louder. If you have time, feel free to test and feedback, this will surely help