Nitrokey/nitrokey-3-firmware #422: Pynitrokey crash on add TOTP code

proninyaroslav, 2023-12-26:
**Device**: Nitrokey 3A
**Firmware:** v1.6.0-test.20231206, v1.6.0-test.20231218
**Pynitrokey**: 0.4.44

I'm trying to add TOTP code and I get pynitrokey crash:
```
yaroslav@yaroslav-pc:~$ nitropy nk3 secrets add-otp Bitwarden SECRET --kind=TOTP
Command line tool to interact with Nitrokey devices 0.4.44
Please touch the device if it blinks
Critical error:
An unhandled exception occurred
	Exception encountered: SecretsAppException(code=6983/OperationBlocked)
```

all other Nitrokey 3A functions work (FIDO2, GPG).

<details>
  <summary>Log:</summary>
  
```
200        INFO pynitrokey.cli Timestamp: 2023-12-26 18:31:59.310340
200        INFO pynitrokey.cli OS: uname_result(system='Linux', node='yaroslav-pc', release='6.6.7-200.fc39.x86_64', version='#1 SMP PREEMPT_DYNAMIC Wed Dec 13 21:43:37 UTC 2023', machine='x86_64')
200        INFO pynitrokey.cli Python version: 3.12.1
200        INFO pynitrokey.cli Cli arguments: ['nk3', 'secrets', 'add-otp', 'Bitwarden', 'WEINOGVACOF32SOV6ENTHLDJR7NESTB5', '--kind', 'TOTP']
201        INFO pynitrokey.cli pynitrokey version: 0.4.44
202        INFO pynitrokey.cli cryptography version: 41.0.7
202        INFO pynitrokey.cli ecdsa version: 0.18.0
203        INFO pynitrokey.cli fido2 version: 1.1.2
203        INFO pynitrokey.cli pyusb version: 1.2.1
203        INFO pynitrokey.cli spsdk version: 1.11.0
453        INFO  libusbsio Loading SIO library: /home/yaroslav/.local/pipx/venvs/pynitrokey/lib64/python3.12/site-packages/libusbsio/bin/linux_x86_64/libusbsio.so
457        INFO  libusbsio HID enumeration[94758603654384]: initialized
457       DEBUG  libusbsio HID enumeration[94758603654384]: device #0: MasterKeys S
457       DEBUG  libusbsio HID enumeration[94758603654384]: device #1: MasterKeys S
458       DEBUG  libusbsio HID enumeration[94758603654384]: device #2: MasterKeys S
458       DEBUG  libusbsio HID enumeration[94758603654384]: device #3: Nitrokey 3
458       DEBUG  libusbsio HID enumeration[94758603654384]: device #4: MX Anywhere 2S
458        INFO  libusbsio HID enumeration[94758603654384]: finished, total 5 devices
463       DEBUG fido2.hid.linux Failed opening device /dev/hidraw0
Traceback (most recent call last):
  File "/home/yaroslav/.local/pipx/venvs/pynitrokey/lib64/python3.12/site-packages/fido2/hid/linux.py", line 98, in list_descriptors
    devices.append(get_descriptor(hidraw))
                   ^^^^^^^^^^^^^^^^^^^^^^
  File "/home/yaroslav/.local/pipx/venvs/pynitrokey/lib64/python3.12/site-packages/fido2/hid/linux.py", line 55, in get_descriptor
    with open(path, "rb") as f:
         ^^^^^^^^^^^^^^^^
PermissionError: [Errno 13] Permission denied: '/dev/hidraw0'
464       DEBUG fido2.hid.linux Failed opening device /dev/hidraw1
Traceback (most recent call last):
  File "/home/yaroslav/.local/pipx/venvs/pynitrokey/lib64/python3.12/site-packages/fido2/hid/linux.py", line 98, in list_descriptors
    devices.append(get_descriptor(hidraw))
                   ^^^^^^^^^^^^^^^^^^^^^^
  File "/home/yaroslav/.local/pipx/venvs/pynitrokey/lib64/python3.12/site-packages/fido2/hid/linux.py", line 55, in get_descriptor
    with open(path, "rb") as f:
         ^^^^^^^^^^^^^^^^
PermissionError: [Errno 13] Permission denied: '/dev/hidraw1'
464       DEBUG fido2.hid.linux Failed opening device /dev/hidraw2
Traceback (most recent call last):
  File "/home/yaroslav/.local/pipx/venvs/pynitrokey/lib64/python3.12/site-packages/fido2/hid/linux.py", line 98, in list_descriptors
    devices.append(get_descriptor(hidraw))
                   ^^^^^^^^^^^^^^^^^^^^^^
  File "/home/yaroslav/.local/pipx/venvs/pynitrokey/lib64/python3.12/site-packages/fido2/hid/linux.py", line 55, in get_descriptor
    with open(path, "rb") as f:
         ^^^^^^^^^^^^^^^^
PermissionError: [Errno 13] Permission denied: '/dev/hidraw2'
464       DEBUG fido2.hid.linux Failed opening device /dev/hidraw4
Traceback (most recent call last):
  File "/home/yaroslav/.local/pipx/venvs/pynitrokey/lib64/python3.12/site-packages/fido2/hid/linux.py", line 98, in list_descriptors
    devices.append(get_descriptor(hidraw))
                   ^^^^^^^^^^^^^^^^^^^^^^
[... 94 lines cut ...]
  File "/home/yaroslav/.local/pipx/venvs/pynitrokey/lib64/python3.12/site-packages/pynitrokey/nk3/secrets_app.py", line 390, in _send_receive_inner
    raise SecretsAppException(status_bytes.hex(), "Received error")
pynitrokey.nk3.secrets_app.SecretsAppException: SecretsAppException(code=6983/OperationBlocked)
789       DEBUG       root listing all connected devices:
901       DEBUG       root :: 'Nitrokey FIDO2' keys
901       DEBUG       root :: 'Nitrokey Start' keys:
919       DEBUG       root :: 'Nitrokey 3' keys
923        INFO  libusbsio HID enumeration[94758603146768]: initialized
923       DEBUG  libusbsio HID enumeration[94758603146768]: device #0: MasterKeys S
923       DEBUG  libusbsio HID enumeration[94758603146768]: device #1: MasterKeys S
923       DEBUG  libusbsio HID enumeration[94758603146768]: device #2: MasterKeys S
923       DEBUG  libusbsio HID enumeration[94758603146768]: device #3: Nitrokey 3
923       DEBUG  libusbsio HID enumeration[94758603146768]: device #4: MX Anywhere 2S
923        INFO  libusbsio HID enumeration[94758603146768]: finished, total 5 devices
953       DEBUG       root /dev/hidraw3: Nitrokey 3 04B27C285CC66156AD228FBADEEC284D
953       DEBUG       root print: --------------------------------------------------------------------------------
953       DEBUG       root print: Critical error occurred, exiting now
953       DEBUG       root print: Unexpected? Is this a bug? Would you like to get support/help?
953       DEBUG       root print: - You can report issues at: https://support.nitrokey.com/
953       DEBUG       root print: - Writing an e-mail to support@nitrokey.com is also possible
953       DEBUG       root print: - Please attach the log: '/tmp/nitropy.log.5jx4umy0' with any support/help request!
953       DEBUG       root print: - Please check if you have udev rules installed: https://docs.nitrokey.com/nitrokey3/linux/firmware-update.html#troubleshooting

```
</details>

daringer, 2024-01-02:
Can you please check if the name `Bitwarden` is not already taken by using `nitropy nk3 secrets list` and passing the correct pin. Starting from v1.6 it is not allowed anymore to directly overwrite a secret, you need to first delete it...

proninyaroslav, 2024-01-02:
@daringer 
No
```
~$ nitropy nk3 secrets list
Command line tool to interact with Nitrokey devices 0.4.44
Please provide PIN to show PIN-protected entries (if any), or press ENTER to skip
Please touch the device if it blinks
Current PIN (8 attempts left): 
Please touch the device if it blinks
No credentials found
```

daringer, 2024-01-02:
okok, then maybe something is weird with the secret... can you please confirm that adding a random TOTP works at all:
```
nitropy nk3 secrets add-otp something ABCDEFGH --kind=TOTP
```
afterwards you should see it using `nitropy nk3 secrets list`

proninyaroslav, 2024-01-02:
Yeah... I remembered that I used `Bitwarden` name on my key. But after some firmware update it was erased. And apparently this created the bug.

```
$ nitropy nk3 secrets add-otp something ABCDEFGH --kind=TOTP
Command line tool to interact with Nitrokey devices 0.4.44
Please touch the device if it blinks
Done
$ nitropy nk3 secrets list
Command line tool to interact with Nitrokey devices 0.4.44
Please provide PIN to show PIN-protected entries (if any), or press ENTER to skip
Please touch the device if it blinks
Current PIN (8 attempts left): 
Please touch the device if it blinks
01. something	Totp/Sha1
```

daringer, 2024-01-02:
mmmh, ok, we aren't aware of any data-loss due to firmware upgrades, but we do extensive tests only between stable releases so worst-case there might be a regression from one test version to another. Please make sure that you have paranoia-level backups of the data on your Nitrokey 3 if you are using `test` firmwares regulary - our test iterations are smaller between `test`-releases.

back to the issue: can you please `nitropy nk3 secrets reset` and then try to re-add the Bitwarden secret you tried initially?

proninyaroslav, 2024-01-02:
Works fine. Thank you.

closed 2024-01-02 by proninyaroslav