Nitrokey/nitrokey-3-firmware #423: Adding TOTP secret fails with exception Error('Incorrect padding')

jplejacq-quoininc-com, 2023-12-27:
device: Nitrokey 3
firmware-version: v1.6.0
nitropy-version: 0.4.44
OS: Debian 11

When adding a TOTP secret for certain sites, amazon.com in particular, nitropy fails with the following error. I've successfully added other TOTP secrets.

$ ./.local/bin/nitropy nk3 secrets add-otp --kind 'TOTP' --hash 'SHA256' --touch-button --protect-with-pin "com.amazon:xxxx@gmail.com" '2PUIUF7EP3OYF3F3CBAMB5QG57XBZMDEMVXYLY57WGIXTKMMIDXQ'
Command line tool to interact with Nitrokey devices 0.4.44
Critical error:
An unhandled exception occurred
        Exception encountered: Error('Incorrect padding')

--------------------------------------------------------------------------------
Critical error occurred, exiting now

I've attached the log file after renaming to allow upload to GitHub.
[nitropy._fn51of8.log](https://github.com/Nitrokey/nitrokey-3-firmware/files/13780685/nitropy._fn51of8.log)

jplejacq-quoininc-com, 2023-12-27:
I found this thread in the support forum that does resolve the issue. Is this a firmware or a nitropy issue?

https://support.nitrokey.com/t/nk3-mini-unable-to-register-otp-secret/5052

daringer, 2024-01-02:
yes, this is a client side issue - there already is an issue in pynitrokey: https://github.com/Nitrokey/pynitrokey/issues/374
closing in favor of this, thanks for reporting!

closed 2024-01-02 by daringer