Nitrokey/nitrokey-3-firmware #424: Adding HOTP Secret via heads factory reset/gui fails since Update to v1.6 (any Version)

Oessel, 2024-01-01:
After updating my NK 3 A NFC to v1.6 (all Test and stable Versions are affected)  it is Not possible to perform a factory reset with heads or to Set a new HOTP Secret via GUI. After scanning the QR Code for the TOTP Secret i get prompted to enter my NK3 admin pin, after that i get "setting HOTP Secret on Nitrokey failed!" Of course i tried it a couple Times and with different Versions of heads but it's Always the same result.

sosthene-nitrokey, 2024-01-02:
Hi, Thank you for the report.

We are aware of the issue and a fix is in the process of being published. See https://github.com/Nitrokey/nitrokey-hotp-verification/pull/28 for more details.

daringer, 2024-01-02:
@Oessel to add to this, there will be a new nitropad release with heads today/tomorrow, which will fix this.
On top, you can by now also workaround this issue by deleting the HOTP secret via another machine using pynitrokey.

daringer, 2024-01-02:
to be more precise, this should be solved here: https://github.com/Nitrokey/heads/tree/nitropad-release-v2.4
closing this issue, please re-open if you encounter the same issue after updating to v2.4

closed 2024-01-02 by daringer