Nitrokey/nitrokey-3-firmware #440: Avoid is_mountable calls

robin-nitrokey, 2024-02-13:
[`Filesystem::is_mountable`](https://docs.rs/littlefs2/latest/littlefs2/fs/struct.Filesystem.html#method.is_mountable) calls `Filesystem::mount`, checks the result and drops the mounted filesystem.  This means that code that first calls `is_mountable` and then `mount` on the same storage mounts the storage twice.

https://github.com/Nitrokey/nitrokey-3-firmware/blob/6f032b5733b9c40621069cb29ef00bed028863fd/components/boards/src/store.rs#L275-L283

Instead we should first call `Filesystem::mount` and handle the error case directly.

robin-nitrokey, 2024-04-24:
This should significantly reduce bootup time.  With logging enabled (`log-semihosting,log-info`), removing the `is_mountable` call reduces the mount time from 1 000 ms to 500 ms and the total bootup time from 2 300 ms to 1 800 ms.  Unfortunately, the borrow checker does not allow us to re-try the mount in the error case because of the static lifetime of the returned filesystem.  Maybe we can find a workaround.

sosthene-nitrokey, 2024-04-25:
We could have a `mount_or_format` method in `littlefs2` ?

I'm not sure I see the borrow checker error. Is this [`NLL case N°3`](https://rust-lang.github.io/rfcs/2094-nll.html#problem-case-3-conditional-control-flow-across-functions) again?

robin-nitrokey, 2024-04-25:
Yes, we can do it in `littlefs2` but not directly in the firmware.

closed 2024-10-25 by robin-nitrokey