Nitrokey/nitrokey-3-firmware #447: nrf-builder does not work with `path` patch dependencies

sosthene-nitrokey, 2024-02-28:
For some reason the gen-command-bd started failing because from the nrf-builder directory the relative path to a patch dependency is invalid. Not sure why it stopped working only now?

https://github.com/Nitrokey/nitrokey-3-firmware/blob/7b12be5cc393a52c810056087c9d6b0ee1262c94/utils/nrf-builder/Makefile#L29

robin-nitrokey, 2024-02-28:
Probably caused by the introduction of the nkpk runner (https://github.com/Nitrokey/nitrokey-3-firmware/commit/a2639f0449d2ecd7db97dfccf70c83fce5c7ae4) because we now parse the Cargo manifest of the runner to determine the version instead of just depending on the `utils` component.

sosthene-nitrokey, 2024-02-29:
I found the issue. Instead of using `../..`, `nrf-builder` uses a soft link to `../..` in `utils/nrf-builder/nk3-fw`, this causes `gen-command-bd` to run with the root being `nitrokey-3-firmware/utils/nrf-builder/nk3-fw`, so it tries to look for the relative patch path from there instead of from `nitrokey-3-firmware`.

closed 2024-02-29 by sosthene-nitrokey