Nitrokey/nitrokey-3-firmware #494: Provide an option to pop up desktop notifications on UIF requests

farblos, 2024-04-25:
Another other-smartcard-features, described, for example [here](https://peterbabic.dev/blog/gnupg-pin-cache-smartcards-yubikeys-and-notifications/#yubikey-touch-notification-in-gnome) and having a project [here](https://github.com/maximbaz/yubikey-touch-detector).

I slowly get used to checking the UIF request indicator light, but getting it right onto the desktop would be better.  (Actually, I do not use a full DE, so a more generic norification mechanism I could hook into would be even more appreciated.)

robin-nitrokey, 2024-04-26:
I see that this is a useful feature, but from my perspective, this should be implemented by client applications like gnupg/scdaemon.  The applications have all required information to show such a prompt while tools like `yubikey-touch-detector` can only try to observe side effects.

farblos, 2024-04-26:
Thanks, I thought that was something like an extra feature of the Yubikey.

Uh, the [detection of OpenPGP UIF requests](https://github.com/maximbaz/yubikey-touch-detector#detecting-gpg-operations) done by `yubikey-touch-detector` is really - side-effect-ish?  Will try my luck at the GnuPG mailing list to see whether there are any plans for implementing a cleaner solution.

On the other hand, a clean and general facility to "forward" all NK user presence requests to the desktop, regardless of what (FIDO2, OpenPGP UIF) has caused them, would be probably only feasible for the NK itself to implement, no?

robin-nitrokey, 2024-04-26:
> On the other hand, a clean and general facility to "forward" all NK user presence requests to the desktop, regardless of what (FIDO2, OpenPGP UIF) has caused them, would be probably only feasible for the NK itself to implement, no?

Yes, but this is not trivial.  For applications using the CTAPHID protocol, e. g. FIDO2, we already send out this information as part of the keepalive message.  But this is only possible because it is part of the protocol and can be done with a preliminary response to the command that is being processed. A more generic approach would be possible, but would require a major change to request handling in the firmware.

We will keep this issue as a feature request, but honestly I don’t think we will implement it in the foreseeable future.

farblos, 2024-04-26:
> A more generic approach would be possible, but would require a major change to request handling in the firmware.

I have been half afraid of that.

> We will keep this issue as a feature request, but honestly I don’t think we will implement it in the foreseeable future.

Fair enough, thanks.