Nitrokey/nitrokey-3-firmware #496: ssh-agent cant retrieve key

b90g, 2024-05-09:
I recently updated to 1.7.0 and ever since the ssh-agent doesnt retrieve the resident key

```
[user@disp9643 ~]$ ssh-add -K -vvvv
Enter PIN for authenticator: 
debug3: start_helper: started pid=1276
debug3: ssh_msg_send: type 5
debug3: ssh_msg_recv entering
debug1: start_helper: starting /usr/libexec/openssh/ssh-sk-helper 
debug1: sshsk_load_resident: provider "internal", have-pin
debug1: sk_probe: 1 device(s) detected
debug1: sk_probe: selecting sk by touch
debug1: ssh_sk_load_resident_keys: trying /dev/hidraw0
debug1: check_sk_options: option uv is unknown
debug1: read_rks: existing 2, remaining 8
debug1: read_rks: Device /dev/hidraw0 has resident keys for 2 RPs
debug1: read_rks: rp 0: name="(none)" id="github.com" hashlen=32
debug1: read_rks: rp 1: name="(none)" id="ssh:" hashlen=32
debug1: read_rks: get RKs for /dev/hidraw0 slot 1 failed: FIDO_ERR_PIN_AUTH_INVALID
debug1: ssh_sk_load_resident_keys: read_rks failed for /dev/hidraw0
Provider "internal" returned failure -1
debug1: ssh-sk-helper: sshsk_load_resident failed: invalid format
debug1: main: reply len 8
debug3: ssh_msg_send: type 5
debug1: client_converse: helper returned error -4
debug3: reap_helper: pid=1276
Unable to load resident keys: invalid format
```

intrestingly i used the nitrokey to webauthn/passkey login into github.

generating a new key does not work.

i wonder why reading for RKs it tells me the PIN was invalid, i just get promted once, and the pin is correct. else the credentials wouldnt be shown as far as i understand.

after the the update i did an nk3 test and it passed all tests
i tried to retrieve the key on fedora40&debian12 

should i try anything, need more info?

robin-nitrokey, 2024-05-10:
Release v1.7.0 includes various improvements to the FIDO implementation, including support for PIN protocol 2 and implementing permissions for PIN tokens.  There could be an incompatibility with how these features are handled in the firmware and libfido2.  I’ll try to reproduce the problem.  Which version of libfido2 is installed on your systems?

b90g, 2024-05-10:
Debian:
`libfido2-1/stable,now 1.12.0-2+b1 amd64 [installed,automatic]`
Fedora:
`libfido2.x86_64                                1.14.0-4.fc40                     @System `

fira959, 2024-05-14:
Same problem on Arch:
`Version         : 1.14.0-2`

robin-nitrokey, 2024-05-17:
Thank you for the reports.  I have been able to reproduce the problem.  Indeed we are a bit too strict when validating the permissions on PIN tokens.  This will be fixed in the next firmware release.

ChristianTacke, 2024-05-22:
Thanks to all partipating in this issue! :-)

Note this also affects `fido2-token`:

```console
$ fido2-token -L -k ssh: /dev/hidraw6
Enter PIN for /dev/hidraw6: 
fido2-token: fido_credman_get_dev_rk: FIDO_ERR_PIN_AUTH_INVALID
$ fido2-token -V
1.14.0
```
(fido2-token rebuild from debian/testing.)

P.S.: Since I was too lazy to find this issue, I opened a [thread on the support forum](https://support.nitrokey.com/t/ssh-add-k-not-working-any-longer-after-upgrading-to-1-7-0/6195).

closed 2024-06-04 by robin-nitrokey

ObiWahn, 2024-06-17:
please rename to something containing `fido2 resident storage` i did not find the issue or only very late. This is not about the ssh-agent

b90g, 2024-06-17:
this issues is already closed.

feel free to ask if  https://github.com/Nitrokey/fido-authenticator/issues/80 can be renamed.

robin-nitrokey, 2024-06-20:
This should be fixed in [v1.7.2](https://github.com/Nitrokey/nitrokey-3-firmware/releases/tag/v1.7.2).