Nitrokey/nitrokey-3-firmware #519: Increase Timeout for Reset under Qubes

nestire, 2024-07-22:
Backround is that under Qubes it takes much longer to attach a devices to the relevant qube (3 Clicks) and then execute the reset special if you use the nitrokey-app2 this leaves not much margin. A 20 Second Timeout should be enough

robin-nitrokey, 2024-07-22:
The timeout is defined in `admin-app` and currently set to 15 seconds:

https://github.com/Nitrokey/admin-app/blob/4dd16c02e93b928e722c6d2d1fd301681d0f02df/src/admin.rs#L17

Is a 5 second increase really enough to make a difference?  If we change it, shouldn’t we go for something like 30 seconds directly?

daringer, 2024-08-01:
Oh, right if it's the FIDO2 timeout can we even change it ? 

from the [specs](https://fidoalliance.org/specs/fido-v2.1-ps-20210615/fido-client-to-authenticator-protocol-v2.1-ps-errata-20220621.pdf):
6.6:

> In case of authenticators with no display, request MUST have come to the authenticator within 10 seconds of powering
up of the authenticator.

robin-nitrokey, 2024-08-01:
Yes, that timeout is fixed.

robin-nitrokey, 2024-10-10:
@nestire Can you please confirm that this is about the FIDO2 reset?

robin-nitrokey, 2024-11-04:
Closing as wontfix as we have to comply with the spec for the FIDO2 reset.

closed 2024-11-04 by robin-nitrokey