Nitrokey/nitrokey-3-firmware #526: Nitrokey 3A NFC: GPG key issues and test failure after firmware update to 1.7.2

AnotherStranger, 2024-08-23:
## Description
I'm experiencing issues with my Nitrokey 3A NFC after updating the firmware to 1.7.2. A day after the update, I lost access to my GPG key (all retry counters were set to zero). I thought it was a user error on my side, so I revoked my cert and created a new one. However, the same problem occurred again one day later. I haven't changed my Linux systems in any way and use the key daily for GPG and SSH.

After these problems, I decided to run nitropy nk3 factory-reset and then nitropy nk3 test. However, the test suddenly fails. The test output is attached below.

## Output of `nitropy nk3 test`
```bash
nitropy nk3 test
Command line tool to interact with Nitrokey devices 0.4.50
Found 1 Nitrokey 3 device(s):
- Nitrokey 3 at /dev/hidraw8

Running tests for Nitrokey 3 at /dev/hidraw8

[1/5]   uuid            UUID query                      SUCCESS         F094406B0327BA579C5170253F66F79A
[2/5]   version         Firmware version query          SUCCESS         v1.7.2
[3/5]   status          Device status                   SUCCESS         Status(init_status=<InitStatus.0: 0>, ifs_blocks=81, efs_blocks=478, variant=<Variant.LPC55: 1>)
Running SE050 test: |                                                                                                                                                                                                                                                                                                                        
[4/5]   se050           SE050                           SUCCESS         SE050 firmware version: 3.1.1 - 1.11, (persistent: (32767,), transient_deselect: (607,), transient_reset: (592,))
Please press the touch button on the device ...
[5/5]   fido2           FIDO2                           FAILURE         'x5c'

5 tests, 4 successful, 0 skipped, 1 failed

Summary: 1 device(s) tested, 0 successful, 1 failed

Critical error:
Test failed for 1 device(s)

--------------------------------------------------------------------------------
Critical error occurred, exiting now
Unexpected? Is this a bug? Would you like to get support/help?
- You can report issues at: https://support.nitrokey.com/
- Writing an e-mail to support@nitrokey.com is also possible
- Please attach the log: '/tmp/nitropy.log.plr1w1nm' with any support/help request!
```

## (Maybe) relevant Log Entries
```
2885      DEBUG  fido2.hid Got keepalive status: 02
3136      DEBUG  fido2.hid Got keepalive status: 02
3385      DEBUG  fido2.hid Got keepalive status: 02
3632      DEBUG  fido2.hid Got keepalive status: 02
3884      DEBUG  fido2.hid Got keepalive status: 02
4132      DEBUG  fido2.hid Got keepalive status: 01
4325      ERROR pynitrokey.cli.trussed.test An exception occured during the execution of the test fido2:
Traceback (most recent call last):
  File "/home/andre/.local/pipx/venvs/pynitrokey/lib/python3.10/site-packages/pynitrokey/cli/trussed/test.py", line 153, in run_tests
    result = test_case.fn(ctx, device)
  File "/home/andre/.local/pipx/venvs/pynitrokey/lib/python3.10/site-packages/pynitrokey/cli/trussed/tests.py", line 411, in test_fido2
    cert = make_credential_result.attestation_object.att_stmt["x5c"]
KeyError: 'x5c'
4326      DEBUG       root print: [5/5] fido2           FIDO2                           FAILURE         'x5c'
4326      DEBUG       root print: 5 tests, 4 successful, 0 skipped, 1 failed
4326      DEBUG       root print: Summary: 1 device(s) tested, 0 successful, 1 failed
4326      DEBUG       root print: Critical error:
4326      DEBUG       root print: Test failed for 1 device(s)
4326      DEBUG       root listing all connected devices:
4392      DEBUG       root :: 'Nitrokey FIDO2' keys
4392      DEBUG       root :: 'Nitrokey Start' keys:
4408      DEBUG       root :: 'Nitrokey 3' keys
```
[nitropy.log](https://github.com/user-attachments/files/16687224/nitropy.log)

## Steps to reproduce:
**Disclaimer: I only have one Nitrokey, so I don't know whether this is reproducible**
**Second addition: I also have a Nitrokey Storage 2 as Backup Device. With this device I never had an Issue like that before.**

1. Update Nitrokey 3A NFC firmware to 1.7.2
2. Use the key for GPG and SSH
3. Run `nitropy nk3 factory-reset` and then `nitropy nk3 test`

## Expected behavior:

The test should pass without any errors.

## Actual behavior:

The test fails with a KeyError: 'x5c' error.

## Additional information:

- I've verified that I installed the udev rules.
- The Nitrokey App 2 seems to work just fine.
- I tried this on 2 different Systems (Ubuntu jammy and Fedora 40)
- Nitrokey info:
  ```bash
  nitropy nk3 status
  Command line tool to interact with Nitrokey devices 0.4.50
  UUID:               F094406B0327BA579C5170253F66F79A
  Firmware version:   v1.7.2
  Init status:        ok
  Free blocks (int):  81
  Free blocks (ext):  478
  Variant:            LPC55
  ```

Any Help regarding this Issue is greatly appreciated. Thank you!

fira959, 2024-08-25:
@AnotherStranger Do you have a pin set on your Nitrokey?

I have to user the --pin option to run the tests successfully and they all pass:

```
Running tests for Nitrokey 3 at /dev/hidraw4

[1/5]	uuid     	UUID query              	SUCCESS  	F...20000000000000000
[2/5]	version  	Firmware version query  	SUCCESS  	v1.7.2
[3/5]	status   	Device status           	SUCCESS  	Status(init_status=<InitStatus: 0>, ifs_blocks=208, efs_blocks=471, variant=<Variant.NRF52: 2>)
Running SE050 test: |
[4/5]	se050    	SE050                   	SUCCESS  	SE050 firmware version: 3.1.1 - 1.11, (persistent: (32767,), transient_deselect: (271,), transient_reset: (256,))
Please press the touch button on the device ...
Please press the touch button on the device ...
[5/5]	fido2    	FIDO2                   	SUCCESS  	

5 tests, 5 successful, 0 skipped, 0 failed

Summary: 1 device(s) tested, 1 successful, 0 failed
```

AnotherStranger, 2024-08-26:
Hey @fira959,
I did try this, and unfortunately, this did not change the behavior.
I'm starting to suspect hardware failure because I started to notice that I have to insert the key multiple times before it is recognized (on multiple systems with different OSes).
If it is HW failure, this would be the second NK3 failing on me in the course of one year. So I do really hope that this is not the case. I will write support and link this issue there.

daringer, 2024-08-27:
hey @AnotherStranger , yes this looks like a hardware issue, more precisely there seems to be something broken inside the LPC55s MCU - the error you see indicates that the internal flash has been deleted due to a corrupted filesystem. This even looks like a MCU error. Your device will be replaced through the Nitrokey Support.

closed 2024-08-27 by daringer

AnotherStranger, 2024-08-27:
Hey @daringer,
thanks for taking the time to explain a possible cause!

Would my device be of value for you to check internally?
If so, I will gladly send you the device via mail for your team to inspect.