Nitrokey/nitrokey-3-firmware #528: Cannot compile the firmware error[E0425]: cannot find value `MEMORY_REGIONS` in this scope

yodatak, 2024-09-07:
rustc 1.77.1 (7cf61ebde 2024-03-27)

With git clone form main git
On fedora 31 toolbox 

cargo install flip-link
cargo install cargo-binutils

packages

```
glib2-2.82.0-1.fc41.x86_64
glibc-2.40-3.fc41.x86_64
glibc-all-langpacks-2.40-3.fc41.x86_64
glibc-common-2.40-3.fc41.x86_64
glibc-devel-2.40-3.fc41.x86_64
glibc-gconv-extra-2.40-3.fc41.x86_64
glibc-headers-x86-2.40-3.fc41.noarch
glib-networking-2.80.0-4.fc41.x86_64
llvm-18.1.8-2.fc41.x86_64
llvm-libs-18.1.8-2.fc41.x86_64
clang-18.1.8-3.fc41.x86_64
clang-devel-18.1.8-3.fc41.x86_64
clang-libs-18.1.8-3.fc41.x86_64
clang-resource-filesystem-18.1.8-3.fc41.noarch
clang-tools-extra-18.1.8-3.fc41.x86_64

``` 

```
yodatak@toolbox /v/h/y/P/p/nitrokey-3-firmware (main)> cargo build
warning: Patch `ctap-types v0.2.0 (https://github.com/trussed-dev/ctap-types.git?rev=72eb68b61e3f14957c5ab89bd22f776ac860eb62#72eb68b6)` was not used in the crate graph.
Check that the patched package version and available features are compatible
with the dependency requirements. If the patch has a different version from
what is locked in the Cargo.lock file, run `cargo update` to use the new
version. This may also occur with an optional dependency that is not enabled.
   Compiling admin-app v0.1.0 (https://github.com/Nitrokey/admin-app.git?tag=v0.1.0-nitrokey.13#c2414919)
   Compiling usbd-ctaphid v0.1.0 (https://github.com/trussed-dev/usbd-ctaphid.git?rev=dcff9009c3cd1ef9e5b09f8f307aca998fc9a8c8#dcff9009)
   Compiling secrets-app v0.13.0 (https://github.com/Nitrokey/trussed-secrets-app?tag=v0.13.0#d819e012)
   Compiling fido-authenticator v0.1.1 (https://github.com/Nitrokey/fido-authenticator.git?tag=v0.1.1-nitrokey.19#0fdecc93)
   Compiling trussed-staging v0.3.0 (https://github.com/trussed-dev/trussed-staging.git?tag=v0.3.1#72b08200)
   Compiling embedded-runner-lib v1.7.2 (/var/home/yodatak/Projects/public/nitrokey-3-firmware/runners/embedded)
   Compiling provisioner-app v0.1.0 (/var/home/yodatak/Projects/public/nitrokey-3-firmware/components/provisioner-app)
error[E0425]: cannot find value `MEMORY_REGIONS` in this scope
  --> runners/embedded/build.rs:71:8
   |
71 |     if MEMORY_REGIONS.filesystem.start & 0x3ff != 0 {
   |        ^^^^^^^^^^^^^^ not found in this scope

error[E0425]: cannot find value `MEMORY_REGIONS` in this scope
  --> runners/embedded/build.rs:90:49
   |
90 |     generate_memory_x(&memory_x, template_file, MEMORY_REGIONS);
   |                                                 ^^^^^^^^^^^^^^ not found in this scope

For more information about this error, try `rustc --explain E0425`.
error: could not compile `embedded-runner-lib` (build script) due to 2 previous errors
warning: build failed, waiting for other jobs to finish...

```

sosthene-nitrokey, 2024-09-09:
Hi,

`cargo` is not used as the top-level command for building the firmware since there are multiple versions of the firmware in the repository (nk passkey, nk3am, nk3xn).

You need to use `make` to build the firmware: `make binaries` will build the binaries and store them in `binaries`.

robin-nitrokey, 2024-11-04:
Closing as solved – please reopen if the make targets don’t work for you.

closed 2024-11-04 by robin-nitrokey