Nitrokey/nitrokey-3-firmware #529: Output of secrets get-password --password is not trimmed

orolhawion, 2024-09-10:
I would like to store a password on NK3 to kind of simulate the static password feature of Yubikeys. For this I used the Secrets-App as follows:

`$ nitropy nk3 secrets get-password staticPassword --password`

In order to make it easier to use I create the following alias in my `~/.zprofile`

`alias sp='nitropy nk3 secrets get-password staticPassword --password | pbcopy'`

pbcopy is available on MacOS and makes anything piped to it available on the clipboard.

Unfortunately I cannot use what is in my clipboard now because it has a trailing newline. I would like this to be removed.

daringer, 2024-09-10:
Yes, nitropy is not really fully built for machine readability - especially the final newline is hard to omit, this would finally look pretty bad in regular (CLI) use-cases. But why not remove the trailing newline in your command ? how about:
```
alias sp='nitropy nk3 secrets get-password staticPassword --password | xargs | pbcopy'
``` 

`xargs` should properly remove the newline

orolhawion, 2024-09-10:
Unfortunately it does not. I even tried with `sed` and `awk`.

orolhawion, 2024-09-10:
Does not work ether. `tr -d \\n` does the trick, but removes any whitespace in the string. However, that is ok for my usecase.

closed 2024-09-10 by orolhawion