Nitrokey/nitrokey-3-firmware #535: Add configuration: `opcard.enabled` and `piv.enabled`

sosthene-nitrokey, 2024-10-01:


daringer, 2024-10-02:
This should also gracefully handle NK3s which already use OpenPGPCard, means I'd say that if OpenPGPCard has already been used (initialized) then the `piv.enabled` should be `false` - in any other case both should be `true`.

This way we can be sure that the behavior for OpenPGPCard users doesn't change, right?

daringer, 2024-10-02:
in fact we have also a mechanism in place to inform users during the update and enforce a minimal pynitrokey version ..... which is btw not implemented for nkapp2 - guess this might increase the prio on this xd

robin-nitrokey, 2024-10-02:
We also need to update the mechanism for the Nitrokey SDK.

robin-nitrokey, 2024-10-17:
Done in https://github.com/Nitrokey/nitrokey-3-firmware/pull/539.