Nitrokey/nitrokey-3-firmware #537: FIDO test releases fails to authenticate

jplejacq-quoininc-com, 2024-10-01:
device: Nitrokey 3C
firmware: v1.7.2-test.20240813
OS: Debian Bookworm
Browsers: Firefox 131.0b8, Chrome 129.0.6668.58

After installing the test firmware, I can no longer authenticate to any website using either Firefox or Chrome. I tried registering the Nitrokey which was successful but still fails to authenticate.

At a site:
1. Enter user name and password.
2. Enter pin for Nitrokey.
3. Prompted to touch Nitrokey. It is blinking.
4. Authentication fails. A variety of messages but no useful information.

This is a show stopper

daringer, 2024-10-02:
hey, thanks for reporting!

Reproducing this was possible for us using firefox - but not chrom(ium). 
For me I tried the following combinations using webauthn.io + chromium 129, firefox 131:
* firefox register (ok), firefox auth (fail - also with weird pin prompt, even with UV set to discouraged)
* firefox register (ok), chromium auth (ok)
* chromium register (ok), chromium auth (ok)

So, could you please crosscheck using webauthn.io if this is also true for you?

Why I am asking is because it looks like we've already found and fixed the issue:
* https://github.com/Nitrokey/fido-authenticator/issues/98
* https://github.com/Nitrokey/fido-authenticator/pull/99

At least this change will make authentication work again for me on firefox, so the question remains whether we see something else (on-top?) here, as you reported that the chrome behavior is identical. We'll include the change into the next test release (likely next week), using that you could then confirm if this fixes it for you.

jplejacq-quoininc-com, 2024-10-02:
Here's my results at webauthn.io: These are all on Debian Bookworn.

| browser                 | registration         |  authentication |
| ------------------------- | ----------------------- | ---------------------------------------- |
| Firefox (131.0b8)  | Success                | Failure "The request is not allowed by the user agent or the platform in the current context, possibly because the user denied permission." |
| Chrome  (129.0.6668.58) | Success (but initally failed because device already register. I'm not sure how I got past this)| Success got to logged in page. |
| Chromium (120.0.6099.224) | Success same as chrome | Success got to logged in page. |
| Edge (129.0.2792.50) | Success | Success |
| Opera (113.0.5230.86) | Success | Success |

However, on real world sites like google.com, gandi.net, ... I'm able to register the key but authentication fails in all browsers.

daringer, 2024-10-02:
great, thx for the detailed info - I assume you did not change anything inside the options on webauthn.io ?

jplejacq-quoininc-com, 2024-10-02:
Hi Markus, I did fool around the the advanced options when I had registered the Nitrokey but couldn't authenticate in Firefox. This was blocking me from registering the key in other browsers. I thought setting "Use Security Key" since google recently is forcing passkey might make a difference but I didn't see any. I reverted back.

I think the main take points to take away are:
1. There's an issue with authentication in Firefox in most (all sites).
2. There's an issue with authentication on some sites in all browsers.

daringer, 2024-10-02:
weird - for me chromium works just fine registering + authenticating with gitlab, github and gandi .... but as expected (due to the webauthn.io behaviro) they all don't work with firefox (but they work again with the fix linked above)...

did you see this: 
> did you do a `fido2 reset` while running the test-firmware ?

because the next possibility to fully reproduce your behavior would be resetting on a 1.7.2 stable, then update to the test firmware. Or did you by any chance downgrade to 1.7.2 in between ?

jplejacq-quoininc-com, 2024-10-02:
Here's the steps I took:

1. I had upgraded to 1.7.2 stable and used this for a week or two. No issues.
2. I then updated to v1.7.2-test.20240813 test to enable PIV. This is when the issues started.

I did not do a fido2 reset. I did not see this. I assume this will wipe all my existing credentials which will be very challenging since I cannot get a list of all the sites from the key - a very frustrating factor of the fido protocol.

BTW, another data point. OpenSSH fido based authentication continues to work with the test firmware.

daringer, 2024-10-02:
ok, then the only path we didn't go to reproduce is to have `fido2 reset` being run while on v1.7.2 firmware, then update to the test firmware, then check if authentication after register doesn't work ...

make sure to not downgrade from the test release to the stable release, this particular "downgrade" path will invalidate your fido2 credentials - also please keep in mind that the test version is not intended for production use - be sure to have backup-methods for using your designated services.

Maybe to explain in short what happened here: we implemented ctap2.2 and it looks like firefox is not precisely behaving as the specification suggests (at least how we understand it). So during attestation (authentication) an additional field is returned in the data-map from the nk3 (this is the bug mentioned above which was fixed already and shows correct behavior for me and firefox). Firefox should also ignore this field (according to spec) but doesn't...

but as already implied, there could be another issue, based on your reports - but it's super weird why webauthn.io is behaving so differently for you.

could you maybe do the tests above again with webauthn.io (firefox + chromium is ok), but set all drop downs in "more-options" to "discouraged" ?

jplejacq-quoininc-com, 2024-10-04:
Well, this seems is even odder.

| browser                 | registration         |  authentication |
| ------------------------- | ----------------------- | ---------------------------------------- |
| Firefox (131.0b8)  | Failure (default options) Success (discouraged) | NA (default) Failure (discouraged) |
| Chromium (120.0.6099.224) | Success  with default options or with "discouraged" | Success  with default options or with "discouraged"   |

I use to be able to register with Firefox.

daringer, 2024-10-08:
ok, then let's wait on the upcoming test release and redo some tests - thx for your investigations

sosthene-nitrokey, 2024-10-24:
Hi, We have released [v1.7.2-test.20241022](https://github.com/Nitrokey/nitrokey-3-firmware/releases/tag/v1.7.2-test.20241022). Can you test if it fixes your issue?

jplejacq-quoininc-com, 2024-10-25:
I upgrade to this version from version v1.7.2-test.20240813.

The updated version does work with Firefox (132.0b7) when authenticating 
to Google Workspace using FIDO2.

Thank you for the quick fix. You can close this issue.

closed 2024-10-26 by daringer