Nitrokey/nitrokey-3-firmware #540: Does HMAC-SHA1 Challenge Response work via NFC?

keepassium, 2024-10-08:
We have a few users [asking](https://github.com/keepassium/KeePassium/issues/339) for KeePassXC-style challenge-response in KeePassium, so I wanted to check how many pieces are missing/already in place.

From a quick search, [it seems](https://github.com/Nitrokey/nitrokey-3-firmware/issues/281) that:
- YubiKey-like protocol was done specifically for KeePassXC
- NC3 firmware must be custom-built with `WITH_XC_YUBIKEY` flag (that is, not available out-of-the-box)
- Communication happens over `pcscd`, a desktop-only daemon. 
- I assume the setup also requires `pynitrokey` as a helper, but am not sure about the general setup.

On iOS, we have very restricted USB and no custom daemons. NFC is pretty much the only viable channel; this is how we work with the YubiKey. 

Considering the current implementation is so focused on desktop and there are [some NFC issues](https://github.com/Nitrokey/nitrokey-3-firmware/issues/398) in general — does NC3 even perform challenge-response via NFC?

robin-nitrokey, 2024-10-08:
Do you want to implement the same mechanism as KeePassXC or would you be open for alternatives?  What protocol do you use with YubiKeys?

For new implementations, we would recommend to use the [FIDO2 hmac-secret extension](https://fidoalliance.org/specs/fido-v2.2-rd-20230321/fido-client-to-authenticator-protocol-v2.2-rd-20230321.html#sctn-hmac-secret-extension).  It is a generic protocol that would also work for other FIDO2 devices and it is easier to set up for end users.  I would also assume that FIDO2 communication is easier to set up than CCID, maybe even over USB, though there might be OS-specific limitations.

Theoretically, the mechanism used for KeePassXC should also work over NFC, but honestly this is nothing we have tested or optimized for, so I don’t know how well it would work in practice.

keepassium, 2024-10-08:
> Do you want to implement the same mechanism as KeePassXC or would you be open for alternatives? 

Same mechanism, it has to be compatible with KeePassXC. Pioneering a different-but-better protocol, with an uphill battle for early adopters is certainly out of scope.

> What protocol do you use with YubiKeys?

I cannot give a more precise answer than ISO7816-4. Over NFC, YubiKey acts as an ISO7816-4 smart card. KeePassium forms and parses raw APDUs ([like this](https://github.com/keepassium/KeePassium/blob/4d2931c807d20ad23716fdd749426f0f0d181e12/KeePassium/challenge-response/ChallengeResponseManager.swift#L423)), whereas session control and actual communication are handled by [YubiKit](https://github.com/Yubico/yubikit-ios/) (as a relatively thin wrapper around iOS' [CoreNFC](https://developer.apple.com/documentation/corenfc)). 

So the ultimate question is whether we might make it work by changing a few bytes in the APDUs (and possibly YubiKit), or it would require a completely different implementation.

> Theoretically, the mechanism used for KeePassXC should also work over NFC, but honestly this is nothing we have tested or optimized for, so I don’t know how well it would work in practice.

Can you please confirm whether this hypothetical statement applies to default/factory firmware?

robin-nitrokey, 2024-10-08:
> So the ultimate question is whether we might make it work by changing a few bytes in the APDUs (and possibly YubiKit), or it would require a completely different implementation.

As far as I see, the same APDUs should work with the Nitrokey 3 out of the box.  I don’t know about the low-level code.

> Can you please confirm whether this hypothetical statement applies to default/factory firmware?

Yes, the challenge-response mechanism for KeePassXC is enabled by default in all recent firmware versions.

robin-nitrokey, 2024-10-08:
Oh, wait, I missed an important detail – the storage chip required for the application that provides the challenge-response mechanism for KeePassXC is not available over NFC, so it is not only unlikely but impossible that it would work.  So FIDO2/hmac-secret is indeed the only option to realize this over NFC.

keepassium, 2024-10-08:
Thank you, @robin-nitrokey, this answers the question. I believe we can close here.

closed 2024-10-08 by keepassium