Nitrokey/nitrokey-3-firmware #553: Implement a mechanism for device authentication

robin-nitrokey, 2024-10-29:
Before sending PINs or other sensitive data to the device, users might want to ensure that the device has not been tampered with (e. g. exchanged or reset).  This could be implemented by signing a nonce or a hash of the configuration or state with the Trussed device key.

*Please upvote :+1: this issue if you would like to see this feature implemented.*

qwrd, 2024-10-30:
I take it this would have to be designed in a way where the user would have to manually perform this verification i.e. via command line, or via the nk-app2? Is it even possible to do in an automated fashion? Maybe by the nk-app2 upon device insertion? But there are multiple PIN's available for different functions of the Nitrokey 3. Suppose a user plugs in NK3 on a device with only PGP support, and enters the PIN to perform some email cryptography. What then?

robin-nitrokey, 2024-10-30:
Yes, the idea is that our tooling (pynitrokey, Nitrokey App 2) would check it when connecting to a device.  If we have such a feature, we can see how it can be integrated with other applications – maybe wrapping the relevant tools or integrating it upstream.

hailfinger, 2025-03-22:
This would also be a useful prerequisite for #616 if the key usage counter is part of the signed state.