Nitrokey/nitrokey-3-firmware #556: Support for post-quantum algorithms, roadmap?

Mithrandir2k18, 2024-11-05:
Some post-quantum algorithms have [already been published and standardized](https://github.com/open-quantum-safe/liboqs). I'm interested in buying some Nitrokeys, but if migrating to post-quantum algorithms isn't planned, I'd rather wait for a future version supporting them, as it seems like adoption [is already being pushed for](https://intelligencecommunitynews.com/ic-insiders-nists-quantum-standards-the-time-for-upgrades-is-now/).

daringer, 2024-11-27:
There is no clear roadmap as we are mostly waiting for roadmaps ;) No, seriously the main driver will of course be NIST - as also mentioned in your first link others will design their solutions based on these decisions. Furthermore Nitrokey 3 and its cryptographic operations are relying strongly on host/platform tools: gpg, opensc, fido (webauthn) and so on - not to talk about the OpenPGPCard spec. These have to go the first steps, we will follow fast - in fact we are already working on proof-of-concept implementations - but a roadmap with some reliable in-production target is currently not possible from our point of view.

Mithrandir2k18, 2024-11-28:
Thank you, that makes sense. So since you said that the Nitrokey 3 relies on the host, is it fair to assume that once gpg etc. support them, the Nitrokey 3 (and maybe older ones as well?) will be able to store the new types of keys as well?

daringer, 2024-11-28:
Which hardware will support which PQC mechanisms is still to be determined. Nitrokey 3 is for sure the most capable, but I would also not promise wide-support for it w/o having details which ones shall be realized on the desk. Most (if not all) PQC algorithms/mechanisms get/improve their quantum resistance through sheer size - means the keys grow massively in size therefore it is quite unrealistic that the older Nitrokeys will get PQC support.

Mithrandir2k18, 2024-11-28:
Alright, thank you, that is a risk I'm willing to take. Closing this, since there really isn't anything for you guys to do.

closed 2024-11-28 by Mithrandir2k18

JamarleyHub, 2026-11-02:
Not to necropost or anything, but has your roadmap changed two years after this? Standardization is now much more advanced, roadmaps regarding compliance is even clearer now, the quantum threat more present than ever and GPG also has some limited support for PQC.

sosthene-nitrokey, 2026-11-02:
Hi,

While PQC algorithms themselves are standardized and begin to be deployed, there is still no standardization of smartcard APIs (especially FIDO and OpenPGP Smartcard) for PQC.

We already have a proof of concept of FIDO PQC and there is ongoing work to get to the same state on the OpenPGP side, but this will only be released in the stable firmware once it's standardized and there is host-side tooling for it.