Nitrokey/nitrokey-3-firmware #570: Chrome hangs out on 1.8.0

proninyaroslav, 2024-12-10:
**OS**: Fedora 41
**Browser**: Chrome 131.0.6778.108
**Device**: Nitrokey 3A
**Firmware**: 1.8.0-RC1 and RC2

When I insert the key and try to launch Chrome, the GUI works fine, but loading the URL hangs forever, I can't open any site. At the same time, the Nitrokey indicator glows green. To fix the problem, I take out the key. I have to do this every time I launch Chrome.

robin-nitrokey, 2024-12-10:
Thank you for the report!  Can you please share the Chrome device log (`chrome://device-log/`) after removing the key?  You can also send it to support@nitrokey.com if you don’t want to share it here.

v1.8.0 adds the PIV smartcard application.  Potentially Firefox or Chrome could try to access the application and, for some reason, stall.  If you are not using PIV, can you please try to disable it and see if that fixes the problem?  You can do that with:

```
$ nitropy nk3 set-config piv.disabled true
```

proninyaroslav, 2024-12-10:
> Can you please share the Chrome device log

```
[22:19:27] USB device removed: path=/dev/bus/usb/003/004 guid=2c3d5f0c-3576-45b7-9daf-b6cb806601ac
[22:19:08] USB device added: path=/dev/bus/usb/003/004 vendor=8352 "Nitrokey", product=17074 "Nitrokey 3", serial="", guid=2c3d5f0c-3576-45b7-9daf-b6cb806601ac
[22:19:08] USB device added: path=/dev/bus/usb/001/003 vendor=32903 "", product=50 "", serial="", guid=d6ee94a5-b6f4-4e47-bc3d-c3fd4c51476e
[22:19:08] USB device added: path=/dev/bus/usb/001/002 vendor=9494 "Cooler Master Technology Inc.", product=87 "MasterKeys S", serial="", guid=955a9f0f-8c20-42fe-b173-d8cac41fc522
```

> $ nitropy nk3 set-config piv.disabled true

Yes, it worked. Looks like the reason is PIV.

tominglis, 2024-12-28:
I just updated my Nitrokeys to 1.8.0 and experienced this same issue on Fedora 40 with Chrome and Firefox. Device LED mostly on (green), Firefox very sluggish loading pages, and Chrome unable to load any pages.

I disabled PIV and they appear to be fine now.

robin-nitrokey, 2024-12-30:
Do you see `p11-kit-proxy` in *Settings* → *Privacy & Security* → *Security Devices* in Firefox?  Some users had similar problems and could fix it by removing this module.  This was with OpenPGP only though, not with PIV.