Nitrokey/nitrokey-3-firmware #573: Add Nitrokey identifiers to relevant (Ubuntu) Snaps' udev rules

daringer, 2024-12-14:
We should make sure (and test) that firefox/chromium/... snap users are able to use Nitrokeys out-of-the-box or at least have the required configs (or whatever needed) available. 

potential template for this: https://support.nitrokey.com/t/nitrokey-fido2-and-fido-u2f-dont-work-on-ubuntu/3957/8 - essentially every devices seems to need a "registration" using the `snap-device-helper`.

For the Chromium Snap I found: https://bugs.launchpad.net/snapd/+bug/1940467 - and for me this reads like this should "just work" by now, but we've received reports that (specifically for the nkpk) there is need to add the line (pattern) from above to `/etc/udev/rules.d/70-snap.firefox.rules` - we should verify that it works and maybe document what to do if not...

snapd seems to know us as [u2f-device](https://github.com/canonical/snapd/blob/master/interfaces/builtin/u2f_devices.go#L101-L119)

robin-nitrokey, 2024-12-14:
I think I registered our VID/PID in some snap repository in the very early phase.  I’ll try to find out where.

robin-nitrokey, 2024-12-14:
https://github.com/canonical/snapd/pull/10937

closed 2025-02-11 by daringer