Nitrokey/nitrokey-3-firmware #574: Update to 1.8.0 fails

orolhawion, 2024-12-17:
When I tried to update my NK3 C NFC to 1.8.0 the following happend. Please find the log file attached.
```log
$ nitropy nk3 update
Command line tool to interact with Nitrokey devices 0.7.0
Do you want to download the firmware version v1.8.0? [Y/n]: y
Download v1.8.0: 100%|███████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 1.08M/1.08M [00:00<00:00, 27.0MB/s]
Current firmware version:  v1.7.2
Updated firmware version:  v1.8.0

Please do not remove the Nitrokey 3 or insert any other Nitrokey 3 devices during the update. Doing so may damage the Nitrokey 3.
Do you want to perform the firmware update now? [y/N]: y

Please press the touch button to reboot the device into bootloader mode ...

Perform firmware update: 100%|█████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 548k/548k [00:06<00:00, 86.2kB/s]
Finalize upgrade:  11%|█████████████████▍                                                                                                                                             | 11/100 [00:08<01:05,  1.35%/s]
Critical error:
An unhandled exception occurred
	Exception encountered: OSError('Failed reading a response')

--------------------------------------------------------------------------------
Critical error occurred, exiting now
Unexpected? Is this a bug? Would you like to get support/help?
- You can report issues at: https://support.nitrokey.com/
- Writing an e-mail to support@nitrokey.com is also possible
- Please attach the log: '/var/folders/3r/rvk9j5zx4xb810y7ypyt0hhr0000gn/T/nitropy.log.karseqwd' with any support/help request!
--------------------------------------------------------------------------------
```
[nitropy.log.karseqwd.zip](https://github.com/user-attachments/files/18161732/nitropy.log.karseqwd.zip)

However, after unplugging the device and retrying the update was ok.

udf2457, 2025-01-13:
I second this observation. Fails first update, ok the second time.

I seem to recall reading on the Nitrokey forums there was a faulty batch built from "less than reliable" parts during COVID times ?  Maybe faulty updates and faulty batch are related ?

sambonbonne, 2025-01-13:
@udf2457 I have this issue on a 3C I bought on November 2024 (from Nitrokey directly), it make me think it's not related to the faulty batch.

robin-nitrokey, 2025-01-13:
This does not look like a hardware issue to me, more like a race condition or logic error in pynitrokey.  It looks like the update was successful.  The finalization step that fails is basically just waiting for the device to show up so that it is not removed while the update is applied.  I’ll try to find out what could go wrong there.

udf2457, 2025-01-13:
@robin-nitrokey FYI it happens in Nitrokey App 2 too.

orolhawion, 2025-01-14:
> It looks like the update was successful.

If I remember correctly you are right about this. I think before the second update try I checked which version is installed and it said 1.8.0, that I am pretty certain about. However, I repeated the update and got the success message which made me feel better.