Nitrokey/nitrokey-3-firmware #578: C_GetAttributeValue should not return CKR_GENERAL_ERROR

yota9, 2025-01-17:
Hi! Duplicating my [OpenSC](https://github.com/OpenSC/OpenSC/issues/3320) issue here, since I suspect the error is returned from the token firmware.

When trying to get CKA_VALUE of some CKO_DATA objects on token (e.g. X.509 Certificate for Card *, Security Object & etc) C_GetAttributeValue returns CKR_GENERAL_ERROR which I assume is wrong behaviour.
By PKCS11 standardI would expect it to return CKR_ATTRIBUTE_SENSITIVE and CK_UNAVAILABLE_INFORMATION set for ulValueLen of CKA_VALUE.

From the pkcs11 standard for C_GetAttributeValue:

```
1. If the specified attribute (i.e., the attribute specified by the type field) for the object cannot 
be revealed because the object is sensitive or unextractable, then the ulValueLen field in that triple is 
modified to hold the value CK_UNAVAILABLE_INFORMATION.
....
If case 1 applies to any of the requested attributes, then the call should return the value CKR_ATTRIBUTE_SENSITIVE
....
Note that the error codes CKR_ATTRIBUTE_SENSITIVE, CKR_ATTRIBUTE_TYPE_INVALID, and CKR_BUFFER_TOO_SMALL 
do not denote true errors for C_GetAttributeValue. If a call to C_GetAttributeValue returns any of these three values, 
then the call MUST nonetheless have processed every attribute in the template supplied to C_GetAttributeValue. 
Each attribute in the template whose value can be returned by the call to C_GetAttributeValue will be returned by 
the call toC_GetAttributeValue.
```

Thanks!

robin-nitrokey, 2025-01-17:
Thanks for the report!  With which device do you see this problem?  And what OpenSC driver do you use, `openpgp` or `piv`?  Can you share an example operation and the OpenSC log for that operation (e. g. by setting `OPENSC_DEBUG=9`)?

yota9, 2025-01-18:
Hello!
I'm using nitrokey 3c with opensc-pkcs11.so module. I'm using custom tool which lists objects and values simultaneously but it could be reproduced by these commands with pkcs11-tool

```
pkcs11-tool -O
....
Data object 3578401536
  label:          'Security Object'
  application:    'Security Object'
  app_id:         2.16.840.1.101.3.7.2.144.0
  flags:          <empty>
  uri:            
...
```

```
pkcs11-tool --read-object --type data --label "Security Object"

warning: PKCS11 function C_GetAttributeValue(VALUE) failed: rv = CKR_GENERAL_ERROR (0x5)
error: get CKA_VALUE failed
Aborting.
```

Full debug log could be found here https://pastebin.com/f9yCBUHR
Thank you!

robin-nitrokey, 2025-01-18:
Thanks for the details.  So you are communicating with the PIV application.  According to the logs, this is the command that fails:

```
Outgoing APDU (11 bytes):
00 CB 3F FF 05 5C 03 5F C1 06 00 ..?..\._...
Incoming APDU (2 bytes):
6A 82 j.
```

OpenSC requests the Security Object DO and the card returns a 0x6a82 Not Found error.  I think this is the expected behavior for the device because the DO has not been set.  What behavior would you expect instead?

yota9, 2025-01-18:
Well at my point of view since it is in the list of CKO_DATA and CKA_VALUE is part of the DO attribute I cat suggest
1. Not list this object at all 
2. Return empty CKA_VALUE
3.  At least return "not true error" like CKR_ATTRIBUTE_TYPE_INVALID in this case when the object value in reality is not set...

robin-nitrokey, 2025-01-18:
From my understanding these are mostly things that would have to be changed in OpenSC:
1. As far as I see, the data object is not listed but OpenSC has a hard-coded list of data objects that it enumerates: https://github.com/OpenSC/OpenSC/blob/2fee97b35fa8f1b51394540b259768b61e126a15/src/libopensc/card-piv.c#L604
2. OpenSC could return an empty value if it receives a Not Found error.  I’m not sure if the standard allows us to just return empty objects if no object has been set.
3. The device just returns Not Found.  The CKR_GENERAL_ERROR comes from OpenSC.  It could also return a different error code if it receives Not Found.

@sosthene-nitrokey What do you think about this issue?

yota9, 2025-01-18:
I'm not deeply familiar with PIV standard, unfortunately. But familiar with pkcs11 API that PIV is using. And from pkcs11 stand point CKA_VALUE is optional, AFAICT pkcs11 does not forbid creating CKO_DATA object without it. E.g. some of other tokens I have allows it:

```
Found object:
  CKA_ID: unavailable information or invalid attribute
  CKA_CLASS: CKO_DATA
  CKA_LABEL: Test
  CKA_TOKEN: True
  CKA_PRIVATE: False
  CKA_MODIFIABLE: True
  CKA_DESTROYABLE: True
  CKA_COPYABLE: unavailable information or invalid attribute
  CKA_APPLICATION: no value
  CKA_OBJECT_ID (1 bytes):
00000000:  01                                               .
  CKA_VALUE: no value
```

Where no value means ulValueLen is 0.
I might be wrong, but I assume although it is PIV application it must follow pkcs11 standard and not return general error in place where is no error in reality.
Thank you!

dengert, 2025-01-18:
Also see: https://github.com/OpenSC/OpenSC/issues/3320#issuecomment-2599866094 as this is a low priority issue with OpenSC software.

yota9, 2025-01-18:
Thanks @dengert !Seems it is opensc issue and not nitrokey-specific, so I'm closing the issue.
Thank you for your help  @robin-nitrokey, appreciate that!

yota9, 2025-01-18:
Having a chance to ask @robin-nitrokey is there a reason that key generation mechanisms are unspported using opensc pkcs11 library? AFAIK key generation is supported on key e.g. gpg uses it with card-edit - generate command. Is it possible to enable it through pkcs11 mechanisms? Thank you!

dengert, 2025-01-19:
> is there a reason that key generation mechanisms are unsupported using opensc pkcs11?

It depends on the card. As the PIV was designed to be a "read only"  card, i.e. user can not update the card, the issuing agency would do that for the user, using some gov approved card management system,  Only the piv-tool was included to do that based on with is in NIST sp800-73, and was intended only for testing. 

 Since then Yubico, PIVKey from Taglio, Nitrokey and possible others have provided their own tools to provision their own "PIV like" cards with their own extensions, mostly for Windows systems.  Microsoft provides a built in PIV driver which uses   Windows CNG minidrivers and not PKCS11.  Although the OpenSC minidriver can work in windows, it is not enabled by default. I usually tell people with PIV cards on windows to use the card vendor's code or the Microsoft code. OpenSC pkcs11-tool and PKCS11 module are installed and worked with FireFox and Thunderbird which do or did support using PKCS11. But they too are also support using Windows CNG. 

> Also would be great if it would support symmetric algs through pkcs11.. 

OpenSC does, if the card supports it. The intent is OpenSC is designed to do crypto operations on the card, and not to do operations that do not require a card. It does support hash, padding and signature verification. In regards to PIV, It only does RAW RSA, and RAW ECDSA, ECDH on the card, but no symmetric algs on the card.

OpenSC PKCS11 only implements parts of PKCS11, for cards, not a complete PKCS11 implementation.

dengert, 2025-01-19:
In regards to the original problem. I have some ideas which I plan to test using no non Nitro cards to test reading non existing objects.

yota9, 2025-01-20:
Wow thank you so much @dengert for this introduction to PIV, it is really valuable and interesting information for me!
> OpenSC does, if the card supports it. 

Nintrokey supports it, both  AES-128, AES-256  according to specification on the [shop page](https://shop.nitrokey.com/shop/nk3cn-nitrokey-3c-nfc-148#attr=).
The problem with all of the tokens on market I've met is that you've buying "pig in a poke".  I'm not talking specific to Nitrokey, but  for other vendors too. They may claim to support AES and PKCS11 interface. But in fact PKCS11 interface support is limited and "read-only" and not all of the expected interfaces like AES is supported through it. And either you would be able to google it somehow or you need to buy and check it yourself...
The information you provided is a light in darkness for me, at least I'm starting to understand the reasons of such problems :) Although it doesn't resolve the above problem, it would be great if manufactures would have simple table with interfaces and supported mechanisms for it :)

sosthene-nitrokey, 2025-01-20:
> [@sosthene-nitrokey](https://github.com/sosthene-nitrokey) What do you think about this issue

The Nitrokey 3 implements the correct behaviour as specified by PIV. [The standard](https://nvlpubs.nist.gov/nistpubs/SpecialPublications/NIST.SP.800-73-4.pdf) (Part 1 - 3.1.2) says `0x6A82` can be returned meaning "Data object not found"

> Nintrokey supports it, both AES-128, AES-256 according to specification on the [shop page](https://shop.nitrokey.com/shop/nk3cn-nitrokey-3c-nfc-148#attr=).

The algorithm list in the shop does not mean that every algorithm is supported by every application. It list all algorithms supported across all applications, and depending on the application, there may be some limitations to their uses.
Regarding AES and PIV: PIV can only work with AES-256, and only for the administration key (which can't be used for general-purpose cryptography).

yota9, 2025-01-20:
> The algorithm list in the shop does not mean that every algorithm is supported by every application.

Yes, I did understand that. What I don't understand is how I should have guess about it :) Who cares that the token can use AES if it is not supported through the interface I was expected it to have :) Seriously, when I see the list of interfaces and list of mechanisms I suspect that they're all available and usable. E.g. I didn't find exact information about pkcs11 interface and what it supports. It would be great if there would be tables for each of the interface/application and supported alg that would be easily readable and understandable by the clients :)

sosthene-nitrokey, 2025-01-20:
> There is an OpenPGP 4.0 standard , but OpenSC support is based on https://gnupg.org/ftp/specs/OpenPGP-smart-card-application-3.4.0.pdf Google for openpgp.

As far as I'm aware the OpenPGP 4.0 smartcard specification is still being drafted.