Nitrokey/nitrokey-3-firmware #597: Cannot flashing and Provisioning the Device

FahadAaly, 2025-02-25:
I had clone the git repository. I had installed all the packages.
I run the command "make -C runners/embedded build-nk3xn" for compilation and it is successfull.
After compilation I reset the device using "make -C utils/lpc55-builder reset". There are no errors at this stage.
When I run lpc55 ls it shows Bootloader { vid: 1FC9, pid: 0021, uuid: 10B76569862E185887007917C6B9A195 }
so my device is still in bootloader mode.
However, I am going to flash the device using the command "make -C utils/lpc55-builder provision-develop" but it gives me the error 

[image]

[image]

I have no idea what this error is about.

daringer, 2025-02-25:
thanks for reporting, the documentation is not correct there anymore, please use `make -C utils/lpc55-builder provision-release` - once the linked PR is merged this will also be correct in the documentation

FahadAaly, 2025-02-25:
Thanks for the comments. how much time it will take for merge and which branch I can use now?

FahadAaly, 2025-02-25:
@daringer  Another problem is that when I run the command make -C utils/lpc55-builder provision-release. It runs and flash my device but end with error Device not found 42b2. It is searching for the device 42dd. How to reflash now if device is not supporting 42bb?

FahadAaly, 2025-02-25:
@robin-nitrokey this is the commit https://github.com/Nitrokey/nitrokey-3-firmware/issues/597#issuecomment-2682482213

robin-nitrokey, 2025-02-27:
@FahadAaly Again, the full output of `make -C utils/lpc55-builder provision-release` would make it easier to diagnose the problem.  Are the required udev rules in place as described in https://github.com/Nitrokey/nitrokey-3-firmware/blob/main/docs/lpc55-quickstart.md#preparing-the-device?  What is the output of `getfacl /dev/hidraw0`?

FahadAaly, 2025-02-28:
[image]

[image]

[image]

@robin-nitrokey these are the output I am getting

robin-nitrokey, 2025-02-28:
@FahadAaly That looks like a permissions error.  What operating system/distribution do you use?  Did you install the Nitrokey udev rules as described [here](https://docs.nitrokey.com/software/nitropy/linux/udev)?

FahadAaly, 2025-02-28:
@robin-nitrokey I am using this version 

[image]

Also, I have these udev rules installed 

[image]

robin-nitrokey, 2025-03-03:
@FahadAaly In the `getfacl` output, I see the username `Debian-gdm`, but the shell shows the user name `fahad`.  Do you use multiple accounts on the machine?  What is the output of `loginctl list-sessions`?

bamontales, 2025-08-06:
On my end here is what I'm experiencing. 

Issues with resetting and provisioning.

Running `make -C utils/lpc55-builder reset`.

```bash
> make -C utils/lpc55-builder reset
make: Entering directory '/home/user/Documents/nitrokey-3-firmware/utils/lpc55-builder'
./scripts/boot-to-bootrom.sh
Bus 002 Device 015: ID 20a0:42b2 Clay Logic Nitrokey 3
Command line tool to interact with Nitrokey devices 0.7.0
Please press the touch button to reboot the device into bootloader mode ...
./scripts/usbwait.sh 1fc9:0021 20a0:42dd
Waiting for USB device: 1fc9:0021 20a0:42dd .Bus 002 Device 016: ID 20a0:42dd Clay Logic USB COMPOSITE DEVICE
make bl-erase-all
make[1]: Entering directory '/home/user/Documents/nitrokey-3-firmware/utils/lpc55-builder'
lpc55 provision ./config/flash-erase-all.toml
cmd: EraseFlashAll

thread 'main' panicked at /home/user/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/lpc55-0.2.1/src/bootloader/protocol.rs:150:21:
Unknown(10001)
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
make[1]: *** [Makefile:88: bl-erase-all] Error 101
make[1]: Leaving directory '/home/user/Documents/nitrokey-3-firmware/utils/lpc55-builder'
make: *** [Makefile:76: reset] Error 2
make: Leaving directory '/home/user/Documents/nitrokey-3-firmware/utils/lpc55-builder'
```

Running `make -C utils/lpc55-builder provision-release`.

```bash
> make -C utils/lpc55-builder provision-release
make: Entering directory '/home/user/Documents/nitrokey-3-firmware/utils/lpc55-builder'
# TODO: add secure boot
# Step 0: reset CMPA, erase firmware
./scripts/boot-to-bootrom.sh
Bus 002 Device 016: ID 20a0:42dd Clay Logic USB COMPOSITE DEVICE
./scripts/usbwait.sh 1fc9:0021 20a0:42dd
Waiting for USB device: 1fc9:0021 20a0:42dd .Bus 002 Device 016: ID 20a0:42dd Clay Logic USB COMPOSITE DEVICE
make bl-config-cmpa-empty
make[1]: Entering directory '/home/user/Documents/nitrokey-3-firmware/utils/lpc55-builder'
lpc55 configure factory-settings ./config/cmpa-empty.toml

thread 'main' panicked at /home/user/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/lpc55-0.2.1/src/bootloader/protocol.rs:114:14:
success: HidApi(HidApiError { message: "Connection timed out" })
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
make[1]: *** [Makefile:96: bl-config-cmpa-empty] Error 101
make[1]: Leaving directory '/home/user/Documents/nitrokey-3-firmware/utils/lpc55-builder'
make: *** [Makefile:38: provision-release] Error 2
make: Leaving directory '/home/user/Documents/nitrokey-3-firmware/utils/lpc55-builder'
```

Here is my workstation environment.

```bash
Distributor ID:	Linuxmint
Description:	Linux Mint 22.1
Release:	22.1
Codename:	xia
```

Here is my udev rules.

```bash
41-nitrokey.rules  70-lpc55.rules  70-nitrokey-3.rules  70-persistent-cd.rules
```

Nitrokey hardware.

```bash
Bus 002 Device 017: ID 20a0:42b2 Clay Logic Nitrokey 3
```

Here is the `dmesg` log.

```bash
[64115.288120] usb 2-1.1: new full-speed USB device number 17 using ehci-pci
[64115.369834] usb 2-1.1: New USB device found, idVendor=20a0, idProduct=42b2, bcdDevice= 1.07
[64115.369846] usb 2-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[64115.369850] usb 2-1.1: Product: Nitrokey 3
[64115.369853] usb 2-1.1: Manufacturer: Nitrokey
[64115.371817] hid-generic 0003:20A0:42B2.0010: hiddev0,hidraw3: USB HID v1.11 Device [Nitrokey Nitrokey 3] on usb-0000:00:1d.0-1.1/input1
```

robin-nitrokey, 2025-08-06:
@bamontales Just to be sure, you are using a Nitrokey 3 Hacker device and not a regular NK3?

bamontales, 2025-08-06:
> @bamontales Just to be sure, you are using a Nitrokey 3 Hacker device and not a regular NK3?

I'm using Nitrokey 3A NFC.

[image]

robin-nitrokey, 2025-08-06:
@bamontales Regular Nitrokey 3 devices cannot be flashed with custom firmware. You need a [Nitrokey 3 Hacker](https://shop.nitrokey.com/shop/nk3anh-nitrokey-3a-nfc-hacker-404) device for that.

commit a9a934d8 (2025-02-25): utils/lpc: adapt provisioning for a better prince init

fixes #597
commit 26c64145 (2025-03-03): utils/lpc: adapt provisioning for a better prince init

fixes #597