Nitrokey/nitrokey-3-firmware #616: Implement a per-key signing/usage counter

hailfinger, 2025-03-21:
I would like to check how often a specific private key has been used to sign something. This is not a count-down-style key use counter as in https://support.nitrokey.com/t/key-counter-in-nitrokey-hsm/658 , but rather a count-up counter.

Motivation: Allow cross-checks between logs of a local process (step-ca) using the Nitrokey to sign certificates and the number of signatures performed as reported by the Nitrokey. That way, even a rogue admin with physical access to the Nitrokey and the ability to tamper with logs can not conceal that issuing a rogue certificate will have incremented the signature counter.

A similar counter mechanism may also be attractive for Passkey functionality.

It might make sense to sign reports of key usage counts with a device key and ensure freshness of the report.

This is just a nice-to-have wishlist item, not a bug report. Thank you for creating Nitrokeys.

robin-nitrokey, 2025-03-21:
Thanks for the feedback!  This is not a feature that we can provide for the Nitrokey in general, but only for specific applications like OpenPGP, FIDO2/Passkey, PIV, etc.  In which applications would you like to have this feature?

hailfinger, 2025-03-22:
Good question. For my use case, I haven't decided yet if I'll use the PIV application or the OpenPGP application for the intermediate CA in my Smallstep step-ca setup.
According to the Smallstep docs, both applications (PIV and OpenPGP) are supported by step-ca, so for me the deciding factor is where I can get that feature earlier and/or without having to use a GUI, preferably with tools available on a standard Debian install.

robin-nitrokey, 2025-03-23:
OpenPGP already provides a signature counter:

```
$ gpg --card-status | grep "Signature counter"
Signature counter : 7251
```

hailfinger, 2025-03-29:
I didn't know that the signature counter already exists, thank you for explaining.
Is that signature counter cryptographically signed with a way to ensure freshness? Can I ask the owner of a Nitrokey to prove (and not just paste some text output which I have to believe) that the signature counter is indeed current as of now?