Nitrokey/nitrokey-3-firmware #617: Implement a mechanism for limited key export

hailfinger, 2025-03-22:
The Nitrokey HSM 2 has a key export mechanism based on DKEK, but that mechanism has one aspect which is both strength and weakness: You can create an unlimited number of clones.

I would like to propose an alternative which limits the number of key clones: Allow export of a private key only for a select number of public keys of other Nitrokeys (e.g. Trussed device key as mentioned in #553) which have been configured as export targets before the private key was created. That way, n copies of the newly created private key can be encrypted with the respective public keys of the other Nitrokeys and exported in encrypted state. The other Nitrokeys can import the just-generated private key encrypted with their public key. That prevents infinite replication if the other Nitrokeys are trusted not to re-export such an imported key.

The target application would probably be OpenPGP.

This is more of an idea and request for feedback, not even a feature request. Given that Smartcard-HSM firmware development seems to be feature-complete, I thought getting such a feature into the Nitrokey 3 firmware might be easier.

sosthene-nitrokey, 2025-03-28:
While we understand the appeal of such a feature, there are a number of reasons why this will probably not be implemented:

- One of the primary benefit of hardware tokens is to prevent export and duplication of keys
- Such a feature would be an additional attack vector
- OpenPGP keys can be backed up by being generated on the host computer and imported
- Due to technical implementation details, it would not be possible with the `se050` cryptography backend, which is the default and provides additional security as well as improved performance.

hailfinger, 2025-03-29:
Thank you for the excellent explanation.
My hope was to generate the key inside a Nitrokey to for two reasons:
- prevent accidental leakage of the private key
- ensure the quality of the randomness used for key generation.

Based on your guidance, I can get the desired result by generating the key on the host computer, importing it to all designated Nitrokeys, then destroying the storage media of the host computer. Key generation on a use-once offline host computer with a HWRNG is hopefully roughly equivalent to generating the key in a Nitrokey.

closed 2025-03-29 by hailfinger