Nitrokey/nitrokey-3-firmware #634: Add serial number to USB descriptor

robin-nitrokey, 2025-08-08:
We removed the serial number from the USB descriptor so that devices cannot be identified from logs. We should check if this is still required or if we can now show the serial number (potentially depending on a configuration option).

BonusPlay, 2025-08-08:
This would enable `udev` matching to distinguish between multiple nitrokey devices in a single deployment.

hailfinger, 2025-10-05:
Having serial number visibility would be extremely helpful in an environment where multiple keys are being provisioned/used (e.g. in a CA hierarchy) and where you want to make sure not to use the wrong Nitrokey by accident. This also helps in case someone wants to pass specific Nitrokeys to specific VMs running on a host.
That said, for home use (as opposed to server use) it makes sense to raise the bar for attackers by not having a serial number in the USB descriptor and thus in the logs.
A configuration option for a serial number in the USB descriptor (on/off, no need to manually set the serial number) would be highly appreciated.

closed 2026-10-23 by robin-nitrokey