Nitrokey/nitrokey-3-firmware #637: Issues encountered during compilation.

bamontales, 2025-08-29:
Is this normal after executing `make -C utils/lpc55-builder provision-release`?
```bash
mkdir -p data
solo2 pki dev fido data/fido.key data/fido.cert
8940E1C1598732CE3F3D89F85AAD16CA
# TODO: add Trussed key & cert
solo2 app provision store-fido-batch-cert data/fido.cert
Error: Empty list of Solo 2
make[1]: *** [Makefile:123: fw-provision-certs] Error 1
```

And here is the result upon checking using `nitropy nk3 test --exclude provisioner` command.
```bash
Command line tool to interact with Nitrokey devices 0.7.0
Found 1 NK3 device(s):
- Nitrokey 3 at /dev/hidraw3

Running tests for Nitrokey 3 at /dev/hidraw3

[1/5]   uuid            UUID query                      SUCCESS         8DE1FC53654B3A55BCDB7F3D99FB2BF8
[2/5]   version         Firmware version query          SUCCESS         v1.8.2+git.51be6400
[3/5]   status          Device status                   SUCCESS         Status(init_status=<InitStatus.0: 0>, ifs_blocks=62, efs_blocks=478, variant=<Variant.LPC55: 1>)
Running SE050 test: |                                                                                                                                       
[4/5]   se050           SE050                           SUCCESS         SE050 firmware version: 3.1.1 - 1.11, (persistent: (32767,), transient_deselect: (607,), transient_reset: (592,))
Please press the touch button on the device ...
[5/5]   fido2           FIDO2                           FAILURE         'x5c'

5 tests, 4 successful, 0 skipped, 1 failed

Summary: 1 device(s) tested, 0 successful, 1 failed

Critical error:
Test failed for 1 device(s)

--------------------------------------------------------------------------------
Critical error occurred, exiting now
Unexpected? Is this a bug? Would you like to get support/help?
- You can report issues at: https://support.nitrokey.com/
- Writing an e-mail to support@nitrokey.com is also possible
- Please attach the log: '/tmp/nitropy.log.surmjllh' with any support/help request!
- Please check if you have udev rules installed: https://docs.nitrokey.com/nitrokeys/nitrokey3/firmware-update#troubleshooting-linux
```

And I'm getting this issue when running `make -C utils/lpc55-builder flash FEATURES=develop`.
```bash
cargo build --target thumbv8m.main-none-eabi \
        --features board-nk3xn,develop,no-delog, \
         \
        --quiet --profile release
error[E0433]: failed to resolve: could not find `littlefs2_filesystem` in `lpc55_hal`
  --> components/boards/src/nk3xn.rs:37:12
   |
37 | lpc55_hal::littlefs2_filesystem!(InternalFilesystem: (prince::FS_START, prince::BLOCK_COUNT));
   |            ^^^^^^^^^^^^^^^^^^^^ could not find `littlefs2_filesystem` in `lpc55_hal`

error[E0412]: cannot find type `InternalFilesystem` in this scope
  --> components/boards/src/nk3xn.rs:86:33
   |
86 | pub type InternalFlashStorage = InternalFilesystem;
   |                                 ^^^^^^^^^^^^^^^^^^ not found in this scope
   |
help: consider importing this struct
   |
1  + use crate::nk3xn::prince::InternalFilesystem;
   |

warning: unused import: `trussed::types::LfsResult`
  --> components/boards/src/nk3xn.rs:34:5
   |
34 | use trussed::types::LfsResult;
   |     ^^^^^^^^^^^^^^^^^^^^^^^^^
   |
note: the lint level is defined here
  --> components/boards/src/lib.rs:2:24
   |
2  | #![warn(trivial_casts, unused, unused_qualifications)]
   |                        ^^^^^^
   = note: `#[warn(unused_imports)]` implied by `#[warn(unused)]`

Some errors have detailed explanations: E0412, E0433.
For more information about an error, try `rustc --explain E0412`.
error: could not compile `boards` (lib) due to 2 previous errors; 1 warning emitted
```

robin-nitrokey, 2025-09-01:
1. What is the output of `lsusb`, `nitropy nk3 list` and `solo2 ls` before running this step?
2. This is directly caused by (1) failing.
3. I can reproduce the problem, will take a look.

bamontales, 2025-09-01:
Hi @robin-nitrokey,

Kindly see the following results.

```bash
# lsusb
Bus 002 Device 051: ID 20a0:42b2 Clay Logic Nitrokey 3A Mini/3A NFC/3C NFC
```
```bash
# nitropy nk3 list
Command line tool to interact with Nitrokey devices 0.7.0
:: 'NK3' keys
/dev/hidraw0: Nitrokey 3 8DE1FC53654B3A55BCDB7F3D99FB2BF8
```
No result for `solo2 ls`.

robin-nitrokey, 2025-09-01:
I need to check that but the current state of the device should be fine for testing. The only difference is that it also contains the provisioner app that is not required for normal operation.

bamontales, 2025-09-02:
Thanks for that @robin-nitrokey, good insight.

bamontales, 2025-09-02:
> No, this is not normal. A red LED indicates a panic (i. e. a crash) in the firmware. Do you have a debugger?

We only have the `LPCXpresso55S69 LPC55S69-EVK` board.

robin-nitrokey, 2025-09-02:
You can also use that as a debugger, see Section 4.4 of [UM11158](https://www.mouser.com/pdfDocs/UM11158.pdf).

robin-nitrokey, 2025-09-02:
I think you have to run `make -C utils/lpc55-builder run FEATURES=log-error,log-semihosting`. Then you should see a panic message in the output.

robin-nitrokey, 2025-09-02:
Regarding the third problem from the original post: I think you can just run `make -C utils/lpc55-builder flash`. We forgot to update that line in https://github.com/Nitrokey/nitrokey-3-firmware/commit/26c64145525e14cb438ad7ddeef6c96d1e9bc1a4.