Nitrokey/nitrokey-3-firmware #639: v1.8.2 breaks luks without user presence

Meri3252, 2025-09-08:
After updating my nk3 mini to version 1.8.2 I could no longer unlock my luks partition.

using a backup passphrase I went ahead and removed the nk3 from the luks keyslot and enrolled it again with the following steps:

`systemd-cryptenroll --fido2-device=auto --fido2-with-client-pin=yes --fido2-with-user-presence=no --fido2-credential-algorithm=eddsa /dev/sda3`

> 🔐 Please enter current passphrase for disk /dev/sda3: •••••••••••••••••••••••••••••••••
> Initializing FIDO2 credential on security token.
> 👆 (Hint: This might require confirmation of user presence on security token.)
> 🔐 Please enter security token PIN: ••••••••
> Generating secret key on FIDO2 security token.
> 👆 **Got unsupported option error when user presence test is turned off. Trying with user presence test turned on.**
> New FIDO2 token enrolled as key slot 0.

As indicated by the above error, it was not possible to use --fido2-with-user-presence=no with the nk3 using firmware 1.8.2
This has always worked on previous nk3 firmware versions

As a result, any unlocking of luks devices now require both pin and physical touch, which breaks any remote setups.

daringer, 2025-09-15:
> Also related [Nitrokey/nitrokey-hotp-verification#41](https://github.com/Nitrokey/nitrokey-hotp-verification/issues/41)
> 
> Nk3 cannot be used in servers for remote attestation anymore ; touching the device is needed when secret is written into HOTP slot (so ohysical presence + secrets app pin).

No, it is not related - this is about the fido2 application on the Nitrokey 3, hotp is not implemented in fido2, it's part of the secrets app.

daringer, 2025-09-15:
@Meri3252 the main issue we cannot violate the specification is that we are working on the Nitrokey 3 FIDO2 implementation Level 2 certification by the Fido Alliance. I am totally with you, this is from a practical perspective not a user friendly decision for the FIDO2 applicability in every day scenarios. Still, please consider that a user-presence check is indeed an added security layer, as the pin could be automatically injected into the running process on the host - thus for a compromised host this might be a last resort.

Anyways, we are looking into the possibility that this might be configurable, although this would be quite some work to actually make the device less spec conformant. I don't see this happening soon, but @anyone here, please +1 this comment if you need this feature.

Ingo-Albrecht, 2025-09-20:
tl;dr I think the respective implementation to require UP/touch by Nitrokey certainly is an improvement for the user-base and a safer default for regular FIDO2 use-cases.

FIDO2 is a standard to secure user authentication and the main aim of USB token implementing FIDO2 is to keep respective personal secrets portable, i.e. the portability is an intrinsic part of the sec model.

What I miss in this issue report is a disclaimer that using a FIDO2 token to unlock LUKS data-at-rest with a PIN only foregoes the security gained by its portability. It only provides a reasonable assurance, if you have considerable other opsec for the system, e.g. extra physical and changing the PIN each time you use it (back to RFC2289 by foot?).

In other words, it is a use-case, but a very specific one that entails a lot of extra measures to rely on. Once you have implemented said opsec, you might as well deploy other methods systemd already offers, e.g. unlocking LUKS with a token PIV key, tying to a system TPM PCR (if the system has it), or else.

While I truly appreciate the virtue of the heads-project maintainers, heads is designed to provide extra assurance for a measured boot, i.e. the same opsec preconditions apply per se. Its security model measures system integrity with reverse hotp/totp against the system state, relying on the portable secrets being non-tampered. It is not designed to provide a secure channel to enter a PIN on a remote token for the purpose of re-ownership, i.e. resetting the measured boot of the system and token.

Consequently, since an UP protocol requirement is a sensible default for the majority of regular FIDO2 use-cases, it only makes sense if the token defaults to it. If firmware can be extended to be configurable (see note in https://github.com/Nitrokey/nitrokey-3-firmware/issues/532#issuecomment-3313847362) that's great, but disregarding FIDO2 per default helps no user group in my opinion.

Thanks for reading.

hailfinger, 2025-10-05:
If you're willing to physically modify a Nitrokey, you can fake the user presence. With a Nitrokey 3A (not NFC), attaching a large enough piece of aluminium foil to the metal handle may work.