Nitrokey/nitrokey-3-firmware #642: Cant overwrite keys

faljse, 2025-09-17:
```
gpg --edit-key --expert asdf@asdf.at
gpg (GnuPG) 2.4.8; Copyright (C) 2025 g10 Code GmbH
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Secret key is available.

sec  ed25519/AB73B566F2261C22
     created: 2025-09-17  expires: never       usage: SC  
     trust: ultimate      validity: ultimate
ssb  cv25519/159246B9B3DA1568
     created: 2025-09-17  expires: never       usage: E   
ssb  ed25519/921CCD337B5AAE40
     created: 2025-09-17  expires: never       usage: SA  
[ultimate] (1). asdf <asdf@asdf.at>

gpg> keytocard
Really move the primary key? (y/N) y
Please select where to store the key:
   (1) Signature key
   (3) Authentication key
Your selection? 1

gpg: WARNING: such a key has already been stored on the card!

Replace existing key? (y/N) y
```

now i enter the admin pin in the popup
the led then blinks green and immediately turns to red.

No error msg, the led keeps shining red and the program is hanging indefinitely.

cyb3rko, 2025-10-21:
Couldn't overwrite keys, but I could reset the GPG module on v1.8.3 with `factory-reset` and generate a new key.

kanru, 2026-02-08:
Same problem. I first generated a key on my NK3AN, then tried to import a key to overwrite it.

1. Key slot 3 originally had a RSA4096 key I imported
2. I generated a new EDDSA key on the same slot, this went fine
3. I wanted to import another key to the same slot, but it always crashes

```
Firmware version:   v1.8.3
Init status:        ok
Free blocks (int):  28
Free blocks (ext):  461
Variant:            LPC55
```

privat-eco, 2026-09-10:
Same here with 1.8.3.

sosthene-nitrokey, 2026-10-07:
Hi,

I have finally been able to reproduce this, it doesn't affect all key types, which explains why we initially failed to reproduce it. A fix is on the way and should be there in the next release.

closed 2026-10-08 by sosthene-nitrokey